react-native-camera/react-native-camera #585: camera preview not visible when part of Swiper

maddanio, 2017-02-03:
### Steps to reproduce
1. put camera on a swiper card from react-native-swiper along at least one other:

  <Swiper>
     <Camera ...>
     <View...>
</Swiper>

2.

run the app

### Expected behaviour
The preview should show what the camera sees

### Actual behaviour
The preview shows up with a seemingly random color, i.e. black or white in my tests, probably from some other style

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: 1.0.0-alpha, but also with older versions

maddanio, 2017-02-03:
since react-native-swiper seems to be very popular I guess this could be a big issue for anyone wanting to "swipe to camera" like instagram etc.

maddanio, 2017-02-07:
ok, digging now. i just found out that when put into the swiper the camera view seems to be created twice (for whaterver reason). I could imagine that being a problem, since the view is actually created only once, as is the preview layer.

maddanio, 2017-02-07:
interestingly that does not happen to the mapbox view...

maddanio, 2017-02-07:
btw, i am currently blabbering about this both here and on react-native: [https://github.com/facebook/react-native/issues/12262](url)

serranoarevalo, 2017-02-08:
Hey, happened to me too when I was trying to make a snapchat clone. What worked for me was having `loadMinimal={true}` as a prop in the Swiper. 

For example:

`<Swiper ref='swiper' horizontal={false} loop={false} showsPagination={false} index={0} loadMinimal={true}></Swiper>`

maddanio, 2017-02-09:
ok, that works too it seems, thanks!
still leaving this open as it should work anyhow. My plan of triaging it was to make a completely new native view and bit by bit copy code in from the camera view. but now i will postpone that I guess :)

ptomasroos, 2017-02-09:
Why would you want to leave this as open @maddanio ? This feels like it belongs as a problem of the Swiper implementation?

maddanio, 2017-02-09:
Not convinced, since other views have no such problems, including mapbox

SimonErm, 2018-09-14:
This issue is closed because of inactivity.
Furthermore RCTCamera is deprecated now, use RNCamera instead. If this issue is also related to RNCamera, please open a new one.

closed 2018-09-14 by SimonErm

asgerhallas, 2018-11-15:
@maddanio could seem like `<Swiper loadMinimal>` does the trick.