react-native-camera/react-native-camera #587: onFocusChanged and onZoomChanged not called (ios)

lodev09, 2017-02-04:
### Steps to reproduce
1. implement `onFocusChanged` and `onZoomChanged`
2. tap to focus or pinch to zoom the view

### Expected behaviour
Should call these event functions i.e.
```
onFocusChanged={(e) => {
	console.log('focused!');
}}
onZoomChanged={(e) => {
	console.log('zoomed!');
}}
```

### Actual behaviour
Not calling them -- though implementing them does enable tap-to-focus and pinch-to-zoom

### Environment
- **Node.js version**: latest
- **React Native version**: 0.40
- **React Native platform + platform version**:  iOS 10.2 Real device

### react-native-camera
**Version**: master

lodev09, 2017-02-04:
The reason I need these triggered is because I'm going to implement my own custom focused component. Would be nice if we can easily pass a focus component as props but not required.

zawarudo, 2017-02-08:
Can confirm, callbacks are not getting called at all.

ju5td0m7m1nd, 2017-02-15:
It is fixed in the PR https://github.com/lwansbrough/react-native-camera/pull/550,
but still wait for merging

closed 2018-03-26 by lodev09