react-native-camera/react-native-camera #588: Video recording not working from android

subhash8108, 2017-02-06:
Hello,

I have install your component and its working perfectly by camera....

but i was try to video recording from android emulator, but every time crash. video inserted on library but not data transfer... 

I like default video recording. Is it possible to default video recording screen open on my android react native app with your component.

[image]
[image]

My code:
<View style={styles.container}>
	  
        <Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={styles.preview}
		  captureTarget={Camera.constants.CaptureTarget.disk}
		  captureMode = {Camera.constants.CaptureMode.video}
		  type= {Camera.constants.Type.front}
          aspect={Camera.constants.Aspect.fill}>
          <Text style={styles.capture} onPress={this.startRecording.bind(this)}>[Video start Record]</Text>
		  <Text style={styles.capture} onPress={this.stopRecording.bind(this)}>[Video stop Record]</Text>
        </Camera>
      </View>

Please help me....  how to default video recording screen open by your component.

Thanks
Subhash

adamduren, 2017-07-20:
I am experiencing the same issue. I tried using the sample in the readme but setting the captureMode to video. It crashes as soon as I tap the capture button. Here's a stack trace.

```
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470] JNI DETECTED ERROR IN APPLICATION: JNI CallObjectMethod called with pending exception java.lang.RuntimeException: setAudioSource failed.
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at void android.media.MediaRecorder.setAudioSource(int) (MediaRecorder.java:-2)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at java.lang.Throwable com.lwansbrough.RCTCamera.RCTCameraModule.prepareMediaRecorder(com.facebook.react.bridge.ReadableMap) (RCTCameraModule.java:283)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at void com.lwansbrough.RCTCamera.RCTCameraModule.record(com.facebook.react.bridge.ReadableMap, com.facebook.react.bridge.Promise) (RCTCameraModule.java:348)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at void com.lwansbrough.RCTCamera.RCTCameraModule.captureWithOrientation(com.facebook.react.bridge.ReadableMap, com.facebook.react.bridge.Promise, int) (RCTCameraModule.java:512)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at void com.lwansbrough.RCTCamera.RCTCameraModule.access$100(com.lwansbrough.RCTCamera.RCTCameraModule, com.facebook.react.bridge.ReadableMap, com.facebook.react.bridge.Promise, int) (RCTCameraModule.java:38)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at void com.lwansbrough.RCTCamera.RCTCameraModule$2.orientationEvent() (RCTCameraModule.java:496)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at void com.lwansbrough.RCTCamera.RCTSensorOrientationChecker$Listener.onSensorChanged(android.hardware.SensorEvent) (RCTSensorOrientationChecker.java:64)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at void android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(int, float[], int, long) (SystemSensorManager.java:699)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at void android.os.MessageQueue.nativePollOnce(long, int) (MessageQueue.java:-2)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at android.os.Message android.os.MessageQueue.next() (MessageQueue.java:323)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at void android.os.Looper.loop() (Looper.java:136)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:6121)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at java.lang.Object java.lang.reflect.Method.invoke!(java.lang.Object, java.lang.Object[]) (Method.java:-2)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at void com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run() (ZygoteInit.java:889)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:779)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470] 
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]     in call to CallObjectMethod
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]     from void android.os.MessageQueue.nativePollOnce(long, int)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470] "main" prio=5 tid=1 Runnable
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   | group="main" sCount=0 dsCount=0 obj=0x7416af60 self=0xe4e85400
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   | sysTid=935 nice=-4 cgrp=default sched=0/0 handle=0xe7b3e534
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   | state=R schedstat=( 3083325985 2190967293 12600 ) utm=212 stm=94 core=2 HZ=100
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   | stack=0xff72e000-0xff730000 stackSize=8MB
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   | held mutexes= "mutator lock"(shared held)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   native: #00 pc 00350ab5  /system/lib/libart.so (_ZN3art15DumpNativeStackERNSt3__113basic_ostreamIcNS0_11char_traitsIcEEEEiP12BacktraceMapPKcPNS_9ArtMethodEPv+128)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   native: #01 pc 003311f9  /system/lib/libart.so (_ZNK3art6Thread9DumpStackERNSt3__113basic_ostreamIcNS1_11char_traitsIcEEEEbP12BacktraceMap+308)
07-20 11:34:22.572 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   native: #02 pc 00239895  /system/lib/libart.so (_ZN3art9JavaVMExt8JniAbortEPKcS2_+848)
[... cut ...]
07-20 11:34:22.573 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at android.app.ActivityThread.main(ActivityThread.java:6121)
07-20 11:34:22.573 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at java.lang.reflect.Method.invoke!(Native method)
07-20 11:34:22.573 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
07-20 11:34:22.573 935-935/com.awesomeproject A/art: art/runtime/java_vm_ext.cc:470]   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)

```

adamduren, 2017-07-20:
Fixed by adding appropriate permissions in the AndroidManifest.xml. Missed that on the initial scan.

```
  <uses-permission android:name="android.permission.RECORD_AUDIO"/>
  <uses-permission android:name="android.permission.RECORD_VIDEO"/>
  <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
  <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
```

gitlovenotwar, 2017-09-12:
I have the same issue. The video recording on android is broken. It crashes immediately. I've all the required permissions.

FarhanKhalid, 2017-09-19:
+1 to @gitlovenotwar 

For me the error is different:
09-19 17:03:52.430 3252-3252/com.archive E/MediaProfiles: The given camcorder profile camera 0 quality 5 is not found
09-19 17:03:52.430 3252-3252/com.archive E/SensorManager: Exception dispatching input event.
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410] JNI DETECTED ERROR IN APPLICATION: JNI CallObjectMethod called with pending exception java.lang.RuntimeException: Error retrieving camcorder profile params
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at android.media.CamcorderProfile android.media.CamcorderProfile.native_get_camcorder_profile(int, int) (CamcorderProfile.java:-2)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at android.media.CamcorderProfile android.media.CamcorderProfile.get(int, int) (CamcorderProfile.java:398)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at android.media.CamcorderProfile com.lwansbrough.RCTCamera.RCTCamera.setCaptureVideoQuality(int, java.lang.String) (RCTCamera.java:266)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at java.lang.Throwable com.lwansbrough.RCTCamera.RCTCameraModule.prepareMediaRecorder(com.facebook.react.bridge.ReadableMap) (RCTCameraModule.java:262)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at void com.lwansbrough.RCTCamera.RCTCameraModule.record(com.facebook.react.bridge.ReadableMap, com.facebook.react.bridge.Promise) (RCTCameraModule.java:348)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at void com.lwansbrough.RCTCamera.RCTCameraModule.captureWithOrientation(com.facebook.react.bridge.ReadableMap, com.facebook.react.bridge.Promise, int) (RCTCameraModule.java:512)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at void com.lwansbrough.RCTCamera.RCTCameraModule.access$100(com.lwansbrough.RCTCamera.RCTCameraModule, com.facebook.react.bridge.ReadableMap, com.facebook.react.bridge.Promise, int) (RCTCameraModule.java:38)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at void com.lwansbrough.RCTCamera.RCTCameraModule$2.orientationEvent() (RCTCameraModule.java:496)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at void com.lwansbrough.RCTCamera.RCTSensorOrientationChecker$Listener.onSensorChanged(android.hardware.SensorEvent) (RCTSensorOrientationChecker.java:64)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at void android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(int, float[], int, long) (SystemSensorManager.java:481)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at void android.os.MessageQueue.nativePollOnce(long, int) (MessageQueue.java:-2)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at android.os.Message android.os.MessageQueue.next() (MessageQueue.java:323)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at void android.os.Looper.loop() (Looper.java:135)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:5417)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at java.lang.Object java.lang.reflect.Method.invoke!(java.lang.Object, java.lang.Object[]) (Method.java:-2)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at void com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run() (ZygoteInit.java:726)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]   at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:616)
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410] 
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]     in call to CallObjectMethod
09-19 17:03:52.433 3252-3252/com.archive A/art: art/runtime/java_vm_ext.cc:410]     from void android.os.MessageQueue.nativePollOnce(long, int)

FarhanKhalid, 2017-09-20:
Quick update on this: There is no crash when doing this on the device so far (using Nexus 5X, running Android 7.1.2)

gitlovenotwar, 2017-09-20:
Have you tried video recording? For me recording is starting but when you stop the app crashes immediately

FarhanKhalid, 2017-09-21:
Yes I tried it and it worked on the device, will follow up on #887 separately.

fazeelzama20, 2018-03-16:
facing this error when starting video recording
Error: start failed.
    at createErrorFromErrorData (index.bundle:1807)
    at index.bundle:1760
    at MessageQueue.__invokeCallback (index.bundle:2118)
    at index.bundle:1935
    at MessageQueue.__guard (index.bundle:2053)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (index.bundle:1934)
    at debuggerWorker.js:72

drmmyx, 2018-10-14:
[image]
This might be the reason causing it to fail

closed 2019-04-18 by sibelius