react-native-camera/react-native-camera #590: OutOfMemoryError on Android

petegerhat, 2017-02-07:
### Steps to reproduce
1.Open an Android app with `react-native-camera` viewfinder attached to the main screen

### Expected behaviour
The Android app should not crash after loading the screen.

### Actual behaviour
A `java.lang.OutOfMemoryError` is shown with reference to RCTCameraModule

### Environment
- **Node.js version**: v7.2.1
- **React Native version**: 0.39.2
- **React Native platform + platform version**: Android 5.1.1 (compiled for Android 6.0)

### react-native-camera
**Version**: 0.4.0

```
E/AndroidRuntime(30695): FATAL EXCEPTION: main
E/AndroidRuntime(30695): Process: com.reactnativeapp, PID: 30695
E/AndroidRuntime(30695): java.lang.OutOfMemoryError: Failed to allocate a 51121164 byte allocation with 16777120 free bytes and 43MB until OOM
E/AndroidRuntime(30695): 	at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
E/AndroidRuntime(30695): 	at android.graphics.Bitmap.nativeCreate(Native Method)
E/AndroidRuntime(30695): 	at android.graphics.Bitmap.createBitmap(Bitmap.java:946)
E/AndroidRuntime(30695): 	at android.graphics.Bitmap.createBitmap(Bitmap.java:917)
E/AndroidRuntime(30695): 	at android.graphics.Bitmap.createBitmap(Bitmap.java:848)
E/AndroidRuntime(30695): 	at com.lwansbrough.RCTCamera.RCTCameraModule.rotate(RCTCameraModule.java:561)
E/AndroidRuntime(30695): 	at com.lwansbrough.RCTCamera.RCTCameraModule.fixOrientation(RCTCameraModule.java:580)
E/AndroidRuntime(30695): 	at com.lwansbrough.RCTCamera.RCTCameraModule.access$300(RCTCameraModule.java:45)
E/AndroidRuntime(30695): 	at com.lwansbrough.RCTCamera.RCTCameraModule$3.onPictureTaken(RCTCameraModule.java:671)
E/AndroidRuntime(30695): 	at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1165)
E/AndroidRuntime(30695): 	at android.os.Handler.dispatchMessage(Handler.java:102)
E/AndroidRuntime(30695): 	at android.os.Looper.loop(Looper.java:145)
E/AndroidRuntime(30695): 	at android.app.ActivityThread.main(ActivityThread.java:6917)
E/AndroidRuntime(30695): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(30695): 	at java.lang.reflect.Method.invoke(Method.java:372)
E/AndroidRuntime(30695): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1404)
E/AndroidRuntime(30695): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1199)
```

Another log:
```
$ adb logcat *:E
--------- beginning of main
E/mm-camera-AEC(  358): aec_process_pack_output(26):OIS-frame_id(0), exp_index=301, real_gain=2.552688, linecnt=2865 , aec_settled 1, cur_luma(42), lux_idx(293.383881)
...
--------- beginning of system
E/mm-camera-AEC(  358): aec_process_pack_output(49):OIS-frame_id(0), exp_index=301, real_gain=2.552688, linecnt=2865 , aec_settled 1, cur_luma(43), lux_idx(292.875610)
...
F/libc    (14728): Fatal signal 11 (SIGSEGV), code 2, fault addr 0x9c2a009c in tid 14761 (mqt_js)
...
E/mm-camera(  358): cpp_hardware_process_frame:956, v4l2 ioctl() failed. rc:-1, trans_code:-11
E/mm-camera(  358): [cpp_hardware_process_frame:960] drop this frame
E/mm-camera(  358): [cpp_hardware_process_frame:962] cpp drop counter 0
E/mm-camera(  358): cpp_thread_handle_process_buf_event:221] get buffer fail. drop frame id:86 identity:0x20002
...
E/mm-camera-AEC(  358): aec_process_pack_output(110):OIS-frame_id(0), exp_index=301, real_gain=2.552688, linecnt=2865 , aec_settled 1, cur_luma(43), lux_idx(293.525726)
E/        (  333): ro.product_ship = true
E/        (  333): ro.debug_level = 0x4f4c
E/audit   ( 2116): type=1701 msg=audit(1486374184.104:999): auid=4294967295 uid=10256 gid=10256 ses=4294967295 subj=u:r:untrusted_app:s0 pid=14761 comm="mqt_js" reason="memory violation" sig=11
E/WindowState( 1095): getStack: Window{1687cd2 u0 d0 com.google.android.googlequicksearchbox/com.google.android.apps.gsa.searchnow.SearchNowActivity} couldn't find taskId=4195 Callers=com.android.server.wm.WindowState.getDisplayContent:1066 com.android.server.wm.WindowState.isDefaultDisplay:1791 com.android.server.wm.WindowState.getMultiWindowStyleLw:2391 com.android.internal.policy.impl.multiwindow.MultiPhoneWindowManager.shouldEnableLayoutInsetsBySoftInput:1505 
E/WindowState( 1095): getStack: Window{1687cd2 u0 d0 com.google.android.googlequicksearchbox/com.google.android.apps.gsa.searchnow.SearchNowActivity} couldn't find taskId=4195 Callers=com.android.server.wm.WindowState.getDisplayContent:1066 com.android.server.wm.WindowState.isDefaultDisplay:1791 com.android.server.wm.WindowState.getMultiWindowStyleLw:2391 com.android.internal.policy.impl.multiwindow.MultiPhoneWindowManager.shouldEnableLayoutInsetsBySoftInput:1505 
E/mm-camera(  358): cpp_thread_handle_process_buf_event:221] get buffer fail. drop frame id:111 identity:0x20002
E/mm-camera-AEC(  358): aec_process_pack_output(111):OIS-frame_id(0), exp_index=301, real_gain=2.552688, linecnt=2865 , aec_settled 1, cur_luma(43), lux_idx(293.340149)
E/android.os.Debug( 1095): ro.product_ship = true
E/android.os.Debug( 1095): ro.debug_level = 0x4f4c
E/mm-camera(  358): cpp_thread_handle_process_buf_event:221] get buffer fail. drop frame id:112 identity:0x20002
E/mm-camera-AEC(  358): aec_process_pack_output(112):OIS-frame_id(0), exp_index=301, real_gain=2.552688, linecnt=2865 , aec_settled 1, cur_luma(43), lux_idx(293.268280)
E/InputDispatcher( 1095): channel ~ Channel is unrecoverably broken and will be disposed!
E/WindowState( 1095): getStack: Window{1687cd2 u0 d0 com.google.android.googlequicksearchbox/com.google.android.apps.gsa.searchnow.SearchNowActivity} couldn't find taskId=4195 Callers=com.android.server.wm.WindowState.getDisplayContent:1066 com.android.server.wm.WindowState.isDefaultDisplay:1791 com.android.server.wm.WindowState.getMultiWindowStyleLw:2391 com.android.internal.policy.impl.multiwindow.MultiPhoneWindowManager.shouldEnableLayoutInsetsBySoftInput:1505 
[... 175 lines cut ...]
E/mm-camera(  358): mct_pipeline_stop_session: __debug__: X 1792= 
E/mm-camera-img(  358): module_imglib_msg_thread:116] X
E/mm-camera(  358): stats_module_stop_session: 3 port =0xb8973928 name=stats_sink 
E/mm-camera(  358): stats_module_stop_session: __ZZZZ__: X 350= 
E/mm-camera(  358): cpp_module_stop_session:396, info: session 2 stopped.
E/mm-camera(  358): c2d_module_stop_session: __ZZZZ__: E 285= 
E/mm-camera(  358): c2d_module_stop_session:295, info: stopping session 2 ...
E/mm-camera(  358): c2d_module_stop_session:300, info: stopping c2d_thread...
E/mm-camera(  358): c2d_module_post_msg_to_thread:354, msg.type=1
E/mm-camera(  358): c2d_thread_process_pipe_message:453, C2D_THREAD_MSG_ABORT: c2d_thread exiting..
E/mm-camera(  358): c2d_module_stop_session:312, closing c2d subdev...
E/mm-camera(  358): c2d_module_stop_session:334, info: session 2 stopped.
E/mm-camera-img(  358): module_imglib_msg_thread:116] X
E/mm-camera-sensor(  358): actuator_close:820 actuator_close: X
E/mm-camera-sensor(  358): module_sensor_deinit_session: __ZZZZ__: Before unload lib 1066= 
E/mm-camera-sensor(  358): module_sensor_deinit_session: __ZZZZ__: After unload lib 1068= 
E/mm-camera-sensor(  358): module_sensor_deinit_session: __ZZZZ__: X 1084= 
E/mm-camera-sensor(  358): module_sensor_stop_session: __ZZZZ__: X 1233= 
E/mm-camera(  358): mct_controller_destroy: __debug__: X 216= 
E/mm-camera(  358): server_process_hal_event:__DBG__ X sucess event id=2
E/mm-camera(  358): main: __ZZZZ___ Result del session Entry
E/mm-camera(  358): main: __ZZZZ___ Result del session Exit
E/WindowState( 1095): getStack: Window{1687cd2 u0 d0 com.google.android.googlequicksearchbox/com.google.android.apps.gsa.searchnow.SearchNowActivity} couldn't find taskId=4195 Callers=com.android.server.wm.WindowState.getDisplayContent:1066 com.android.server.wm.WindowState.isDefaultDisplay:1791 com.android.server.wm.WindowState.getMultiWindowStyleLw:2391 com.android.internal.policy.impl.multiwindow.MultiPhoneWindowManager.shouldEnableLayoutInsetsBySoftInput:1505 
^C
```

yasintoy, 2017-02-13:
I've just solved this problem by adding "android:largeHeap="true"  to  android/src/main/AndroidManifest.xml. and rebuild your project with react-native run-android

for example:
...
      android:name=".MainApplication"
      android:allowBackup="true"
      android:largeHeap="true"*
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:theme="@style/AppTheme">
...
I've just solved this problem by adding "android:largeHeap="true"  to  android/src/main/AndroidManifest.xml. and rebuild your project with react-native run-android

for example:
...
      android:name=".MainApplication"
      android:allowBackup="true"
      android:largeHeap="true"*
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:theme="@style/AppTheme">
...

ptomasroos, 2017-02-13:
Thanks for the tip @yasintoy

yasintoy, 2017-02-13:
@ptomasroos Your welcome.

dorthwein, 2017-03-09:
maybe close the issue?  I'm evaluating using this and seems great but made me do a double take with how many issues open & the looking for contributors.

npomfret, 2017-03-19:
I've got the same problem on certain devices.  I'll try the suggested fix and report back...

npomfret, 2017-03-19:
Ok, I've deployed and tested on a couple of really old, low memory samsung devices and the crash didn't happen. So I think it looks good.

skovhus, 2017-04-18:
For me this is fixed by reverting to 0.4.0. This https://github.com/lwansbrough/react-native-camera/pull/493 where images are rotated is causing the problem.

You should avoid `largeHeap` (as the Android documentation states)...

npomfret, 2017-04-18:
I _think_ image rotation isn't needed any more as the metadata is preserved in the output image.

skovhus, 2017-04-19:
@npomfret the image rotation for **front facing camera** on 0.4.0 (pre Java rotation introduced in #493) isn't working (at least not on Galaxy S4 and S7). 

I will investigate if this is just because of a math bug in the `adjustCameraRotationToDeviceOrientation` code. If so, all the image mutation can be removed.

skovhus, 2017-04-19:
Even if I hardcode `rotation` in `adjustCameraRotationToDeviceOrientation` for front-facing camera it still doesn't rotate correctly.

skovhus, 2017-04-19:
Verified that this still breaks with master branch. 

I think: Rotation should be done by the native camera API OR if there is enough memory available.

ptomasroos, 2017-04-20:
And otherwise the image should be incorrectly rotated @skovhus ?  I think that will be very hard to understand the behaviour of the library in that case. I agree that it would be beneficial to find a better approach to it but saying we shouldn't rotate images won't work IMHO. Will break anyones expectations

npomfret, 2017-04-20:
Hi. Sorry if I'm asking stupid questions... I'm new to this project.

I don't understand why the rotation is necessary.  I would have thought that if the native camera supplies a byte array with the JPEG image then that image should either be rotated correctly, or contain meta data describing its orientation.  It sounds like in this case that's not happening.

@skovhus can you send an example image that your camera is producing BEFORE any rotation is applied, just a file that contains the `byte[]` supplied by the `onPictureTaken` method.  I'd like to see what data is inside it.  It would be helpful actually to take a couple at different orientations.

skovhus, 2017-04-20:
@ptomasroos Sure. I would love if the library could rotate the image correctly, but we shouldn't to run out of memory trying... That breaks my expectation. : )

@npomfret I ended up downgrading to 0.4.0. I might look more into this further down the road. And then we should export the raw image and analyse it...

skovhus, 2017-04-20:
Created an issues for discussion utilizing third party Android camera libraries https://github.com/lwansbrough/react-native-camera/issues/672

npomfret, 2017-04-24:
@petegerhat are you able to test this PR: https://github.com/lwansbrough/react-native-camera/pull/682

It might help with the OOM error

Wedrex, 2017-10-01:
This issue is still active.

I'm testing with Samsung Galaxy S2 and the second picture makes the app crash. It should work, allthough the phone is old it's still running on Android 4.4 and therefore should be supported.

Rewieer, 2018-02-09:
@Wedrex is it happening due to fixRotation ? Did you disable the fixOrientation prop ?

sibelius, 2018-02-09:
fixed on master with RNCamera

chmielot, 2018-03-03:
I am having the same issue using RNCamera! The app crashes after taking a few images. Sometimes on the 1st one, sometimes on the 5th. Reducing image size doesn't help. Sometimes, when the app doesn't crash, then I see an incomplete image in the preview rendering.

Crash log:
```
03-03 14:37:42.061   543   752 I ImagingSystem: #Camera# [233669690]> 1_startStreaming
03-03 14:37:42.061   543   752 I PpAlgo  : [HWA_CAM3]prepare4StartStreaming() sensor index 4160,3120, 0,0
03-03 14:37:42.062   543   752 I AlgoFilterManager: [HWA_CAM3]KEY_CA_CAM3 takeFilter HAL_STREAM_TINY
03-03 14:37:42.062   543   752 I AlgoFilterManager: [HWA_CAM3]KEY_CA_CAM3 takeFilter HAL_STREAM_PREVIEW
03-03 14:37:42.062   543   752 I CFilterDownscale: virtual TRetCode hw::CFilterDownscale::decideAllocatorProperty(hw::IMemoryAllocator *, IMemoryAllocator::Property &)(123): prop[7, 1440, 1080, 17, 2420003], propResult[7, 1440, 1080, 17, 2420003]!
03-03 14:37:42.062   543   752 I CFilterDownscale: virtual TRetCode hw::CFilterDownscale::decideAllocatorProperty(hw::IMemoryAllocator *, IMemoryAllocator::Property &)(123): prop[7, 1440, 1080, 35, 2420003], propResult[7, 1440, 1080, 35, 2420003]!
03-03 14:37:42.064   543   752 E HISPMsgController: virtual TRetCode hw::CHISPMsgController::registerListener(hw::IMsgListener *): It has registerListener Listener already !
03-03 14:37:42.067   543   752 I ImagingSystem: virtual TRetCode hw::CFilterBase::start()(756): (stream.sink) instance[0x7068448c00].
03-03 14:37:42.067   543   752 I ImagingSystem: virtual TRetCode hw::CFilterBase::start()(756): (stream.sink) instance[0x70685d8000].
03-03 14:37:42.067   543   752 I ImagingSystem: virtual TRetCode hw::CFilterBase::start()(756): (stream.sink) instance[0x7068448800].
03-03 14:37:42.067   543   752 I ImagingSystem: virtual TRetCode hw::CFilterBase::start()(756): (algo.ipp) instance[0x7068505000].
03-03 14:37:42.068   543   752 I ImagingSystem: virtual TRetCode hw::CFilterBase::start()(756): (downscale) instance[0x70659c0f00].
03-03 14:37:42.070   543   752 I ImagingSystem: virtual TRetCode hw::CFilterBase::start()(756): (downscale) instance[0x70659c0a00].
03-03 14:37:42.071   543   752 I ImagingSystem: virtual TRetCode hw::CFilterBase::start()(756): (fork) instance[0x7059df0800].
03-03 14:37:42.071   543   752 I ImagingSystem: #Camera# [233677007]> doAllocate_alloc_Count(4)_WxH(1440x1080)
03-03 14:37:42.071  7631  8784 I art     : Alloc partial concurrent mark sweep GC freed 5132(196KB) AllocSpace objects, 0(0B) LOS objects, 2% free, 147MB/151MB, paused 1.767ms total 46.316ms
03-03 14:37:42.071  7631  8784 I art     : Starting a blocking GC Alloc
03-03 14:37:42.071   543   752 I gralloc : alloc_device_alloc:564: Alloc handle(0x7065834700): interfmt=11, stride=1440, size=2334720, usage=0x2400003
03-03 14:37:42.072   543   752 I gralloc : alloc_device_alloc:564: Alloc handle(0x7065834e00): interfmt=11, stride=1440, size=2334720, usage=0x2400003
03-03 14:37:42.072   543   752 I gralloc : alloc_device_alloc:564: Alloc handle(0x7065869000): interfmt=11, stride=1440, size=2334720, usage=0x2400003
03-03 14:37:42.073   543   752 I gralloc : alloc_device_alloc:564: Alloc handle(0x7065869400): interfmt=11, stride=1440, size=2334720, usage=0x2400003
03-03 14:37:42.073   543   752 I ImagingSystem: #Camera# [233677007]< doAllocate_alloc_Count(4)_WxH(1440x1080)
03-03 14:37:42.074   543   752 I ImagingSystem: virtual TRetCode hw::CFilterBase::start()(756): (algo.ipp) instance[0x7068506800].
03-03 14:37:42.074   543   752 I ImagingSystem: virtual TRetCode hw::CFilterBase::start()(756): (HISP110Filter) instance[0x70684af740].
03-03 14:37:42.076   543  8947 I ImagingSystem: #Camera# [233683772]> doAllocate_alloc_Count(6)_WxH(960x720)
03-03 14:37:42.077   543  8947 I gralloc : alloc_device_alloc:564: Alloc handle(0x7068458f00): interfmt=11, stride=960, size=1040384, usage=0x2400003
03-03 14:37:42.077   543  8949 I ImagingSystem: #Camera# [233685862]> doAllocate_alloc_Count(6)_WxH(614400x1)
03-03 14:37:42.077   543   752 I ImagingSystem: virtual TRetCode hw::CFilterBase::start()(756): (sensor) instance[0x7068494118].
03-03 14:37:42.077   543   752 I ImagingSystem: virtual TRetCode hw::COV13850Spec::getVcmOtpInfo(hwa_af_otp *, int32_t) OTP:otpData->start=0x86, otpData->end=0x1ef
03-03 14:37:42.078   543  8949 I gralloc : alloc_device_alloc:564: Alloc handle(0x706855e900): interfmt=10a, stride=614400, size=614400, usage=0x2400003
03-03 14:37:42.078   543  8947 I gralloc : alloc_device_alloc:564: Alloc handle(0x706855e100): interfmt=11, stride=960, size=1040384, usage=0x2400003
03-03 14:37:42.078   543  8949 I gralloc : alloc_device_alloc:564: Alloc handle(0x706855eb00): interfmt=10a, stride=614400, size=614400, usage=0x2400003
03-03 14:37:42.079   543  8947 I gralloc : alloc_device_alloc:564: Alloc handle(0x706855ed00): interfmt=11, stride=960, size=1040384, usage=0x2400003
03-03 14:37:42.079   543  8949 I gralloc : alloc_device_alloc:564: Alloc handle(0x70684c4100): interfmt=10a, stride=614400, size=614400, usage=0x2400003
03-03 14:37:42.079   543  8949 I gralloc : alloc_device_alloc:564: Alloc handle(0x70684c4300): interfmt=10a, stride=614400, size=614400, usage=0x2400003
03-03 14:37:42.079   543  8947 I gralloc : alloc_device_alloc:564: Alloc handle(0x706855ef00): interfmt=11, stride=960, size=1040384, usage=0x2400003
[... cut ...]
          this.camera = cam;
        }}
        style={styles.preview}
        flashMode={Camera.Constants.FlashMode.off}
        permissionDialogTitle={'Permission to use camera'}
        permissionDialogMessage={'We need your permission to use your camera phone'}
      >
        <TouchableHighlight
          style={styles.capture}
          onPress={this.takePicture.bind(this)}
          underlayColor="rgba(255, 255, 255, 0.5)"
        >
          <View />
        </TouchableHighlight>
      </Camera>
    );
  }
```

Rewieer, 2018-03-03:
Yeah I truly believe allocating 51916812 bytes (412 Mb !) is hell huge. One should never have to process a image this huge on nowadays mobile devices.
Correct me if I'm wrong @sibelius but everywhere I saw a "quality" prop in the Java code it's expected to be a string, not a number.

sibelius, 2018-03-03:
U can fix this using RNCamera with width prop

The problem is with rotating image logic

closed 2018-03-03 by sibelius

chmielot, 2018-03-03:
@sibelius I am using RNCamera and as you can see from the code above I am using the width prop. I am not doing any image rotation. Can you please explain a little bit more in detail?

Rewieer, 2018-03-03:
@chmielot image rotation is done automatically. RNC does this by reading the EXIF data of the image and applying a rotation to it if necessary. But this require reallocating an other bitmap to store the rotated image, hence the crash.

sharmad1205, 2018-06-21:
After updating to 1.1.4 still getting the out of memory error:
1. `Fatal Exception: java.lang.OutOfMemoryError
Failed to allocate a 51121164 byte allocation with 15851520 free bytes and 41MB until OOM
dalvik.system.VMRuntime.newNonMovableArray (VMRuntime.java)
android.graphics.Bitmap.createBitmap (Bitmap.java:915)
com.facebook.imagepipeline.memory.BitmapPool.alloc (BitmapPool.java:55)
com.facebook.imagepipeline.memory.BitmapPool.alloc (BitmapPool.java:30)
com.facebook.imagepipeline.memory.BasePool.get (BasePool.java:260)`

2. `Caused by java.lang.OutOfMemoryError
Failed to allocate a 1050476 byte allocation with 348752 free bytes and 340KB until OOM
java.lang.StringFactory.newStringFromChars (StringFactory.java)
android.util.Base64.encodeToString (Base64.java:456)
org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground (ResolveTakenPictureAsyncTask.java:130)

3. `Caused by java.lang.OutOfMemoryError
Failed to allocate a 51121164 byte allocation with 16776312 free bytes and 46MB until OOM
dalvik.system.VMRuntime.newNonMovableArray (VMRuntime.java)
android.graphics.BitmapFactory.decodeByteArray (BitmapFactory.java:636)
org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground (ResolveTakenPictureAsyncTask.java:83)`

0x01Brain, 2019-03-07:
This is still an issue with version 1.11.1, as I am getting the same exception being thrown by RNC.