react-native-camera/react-native-camera #591: SubView are persisting after transition and camera disappear when going back.

Deevent, 2017-02-08:
Hello,

I'm trying to transition from a view with a camera to another also with a camera. When I push my new screen, the subviews of my first camera are showing up in my second one. And when I go back to my initial screen, there's no more camera.

[image]

ptomasroos, 2017-02-13:
Can you create a repo to reproduce this and I will happily take a look at it

Deevent, 2017-02-13:
Of course ! I created this repo containing my project : https://github.com/Deevent/curly-winner

Thanks for helping !

jamuhl, 2017-03-22:
@Deevent you found a solution for this? we run into the same issue...

Deevent, 2017-03-22:
@jamuhl No still not, I've tried to render null before my transition but the issue is still there. I'm planning to create my own native module (that just displays a camera background) if I don't find a solution by the end of my project.

chillu, 2017-05-30:
Just hit this as well. The [example](https://github.com/lwansbrough/react-native-camera/blob/master/Example/Example.js#L176) doesn't nest components, but rather uses overlays

danyflorez, 2017-08-01:
Did you find a solution for your problem?

closed 2018-09-14 by SimonErm