react-native-camera/react-native-camera #592: React/RCTBridge.h not found

pcooper1979, 2017-02-08:
I've gone through the installation steps and when trying to compile the standard project I get this error, the header search paths are configured as per the installation instructions.

Any idea why?

ptomasroos, 2017-02-08:
Which version of react-native are you running ?

pcooper1979, 2017-02-09:
Version 0.38.0

h124551753, 2017-02-13:
i also meet this problem afer running 'npm install'.
my react-native version: 0.38.0

ptomasroos, 2017-02-13:
You need to upgrade to rn 0.40 and above in order to use packages containing the namespaces React/*.h

cgotcu, 2017-02-19:
I am running with 0.40 (and react 15.4.0) and am just trying to get the Example going but have the same issue as above. I tried a bunch of stuff others have suggested with the same issue:
- removed node_modules and did a fresh npm
- changed the HEADER_SEARCH_PATHS setting to remove react and react-native
- tried both react-native run-ios and XCode for building/running
Any additional thoughts?

mediavrog, 2017-03-16:
+1 running into the same issue as @cgotcu

eqyiel, 2017-05-05:
Related #692

DerJacques, 2017-08-22:
I'm also experiencing this issue on a freshly detached Expo app. However, installing using Pod works fine.

harrisrobin, 2017-10-17:
Also experiencing this.

sibelius, 2018-01-23:
fixed on master

closed 2018-01-23 by sibelius