react-native-camera/react-native-camera #593: App crashes when video recording starts on Sony Xperia Z5 Compact

krillov, 2017-02-09:
### Steps to reproduce
1. Open camera app
2. Try to capture video

### Expected behaviour
Video should be recorded

### Actual behaviour
App crashes directly without any logged errors when the recording with the back camera starts. It works with the front camera.

### Environment
- **Node.js version**: 5.11.0
- **React Native version**: 0.39.0 and earlier
- **React Native platform + platform version**: Android 6.0
- **Device**: Sony Xperia Z5 Compact

### react-native-camera
**Version**: 0.4.1 and earlier

### The app was tested on following devices
-**Samsung Galaxy S4**: OK
-**Samsung Galaxy S5**: OK
-**Samsung Galaxy S5 Neo**: OK
-**Samsung Galaxy S7**: OK 
-**Sony Xperia Z5 Compact**: FAIL

So the app crashes only on the tested Sony and only when using the back camera. Every tested Samsung devices worked fine.

ptomasroos, 2017-02-09:
Do you have a crash log from adb?

krillov, 2017-02-09:
No. I was watching logcat while it happened but it wasn't any error message there. The app just crashed. Problem is I don't have access to a Sony so I can't really test it more.

blaziken105, 2017-02-13:
@krillov I have tried using the example code. The camera works fine and saves the images however the video recorder crashes without logging in both the front and back camera. I tried in on a Samsung Galaxy S5. I believe the error is in the startRecording method of the example, specifically here.

`   this.camera.capture({mode: Camera.constants.CaptureMode.video})
            .then((data) => console.log(data))
            .catch(err => console.error(err));
        this.setState({
          isRecording: 'true'
        });`

The capture method doesn't seem to accept the capture mode as video and crashes for some reason. I can't seem to get any log in the terminal while running it. Any idea what could be wrong?

React Native Version 0.40.0
React Native Camera 0.6.0

krillov, 2017-02-13:
@blaziken105 Sorry, I have no idea. I use exactly the same code as you do. Only difference is I haven't updated to version 0.40 yet. I'm not a JS expert and it has probably nothing to do with the crash but aren't you using string instead of boolean for isRecording?
```
let options = {
  mode: Camera.constants.CaptureMode.video,
  target: Camera.constants.CaptureTarget.cameraRoll
};
this.camera.capture(options)
.then(data => {
  ...
})
.catch(err => {
  console.log(err)
})
```

blaziken105, 2017-02-13:
@krillov seems like a typo, its supposed to be a boolean...

closed 2018-09-14 by SimonErm