react-native-camera/react-native-camera #594: video Callback not working

leonciocaminha, 2017-02-12:
Hello Guys!

when i try this:

    this.state = {
      camera: {
        aspect: Camera.constants.Aspect.fill,
        captureMode: Camera.constants.CaptureMode.still,
        captureAudio: true,
        captureQuality: Camera.constants.CaptureQuality.high,
        captureTarget: Camera.constants.CaptureTarget.cameraRoll,
        type: Camera.constants.Type.back,
        orientation: Camera.constants.Orientation.auto
      },
      isRecording: false
    };

  _takeVideo = () => {
    if (this.camera) {
      this.camera.capture()
          .then((data) => console.log(data))
          .catch(err => console.error(err));
      this.setState({
        isRecording: true
      });
    }
  }

The callback into this.camera.capture works as well BUT if i change 

captureMode: Camera.constants.CaptureMode.video

isnt work anymore, why?

This happen into ios 10 simulator using 
    "react-native": "0.38.0",
    "react-native-camera": "^0.4.1"

Thanks

blaziken105, 2017-02-13:
I have the same issue the only difference is I have updated to react-native 0.40.0

leonciocaminha, 2017-02-13:
@blaziken105 Dude...the callback works! but only into device, in simulator won´t work

blaziken105, 2017-02-13:
I tried it out, doesn't work on the device or the emulator probably a bug within 0.40.0. Could you try upgrading to the latest version and testing if it works? I am using the example code...

ivishnevs, 2017-10-14:
@leonciocaminha did you find out how to resolve this issue?

closed 2018-09-14 by SimonErm