react-native-camera/react-native-camera #597: Example not working

bosung90, 2017-02-14:
### Steps to reproduce
1. clone repo
2. npm install inside Example folder
3. react-native run-android

### Expected behaviour
Example app should work

### Actual behaviour
App crashes with error

### Environment
- **Node.js version**:
7.5.0
- **React Native version**:
0.34.0 (This is what Example project comes in)
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc
Android 6.0.1

### react-native-camera
**Version**: npm version or "master"
master 65f7903

matthewford, 2017-03-23:
anyone else get the example to work with the expo app?

bkotos, 2017-04-01:
Went through the same steps, in addition to running `react-native link react-native-camera`, and it still doesn't work.  It just crashes and the app exits with no error.

jannevickholm, 2017-04-06:
I have the same problem with Expo. Just says `undefined is not an object` and crashes.

bkotos, 2017-04-06:
I followed this tutorial and I was able to get `react-native-camera` to work: https://www.youtube.com/watch?v=pJseuSOC8HA

itinance, 2017-05-03:
`react-native run-ios` gives 

> Print: Entry, ":CFBundleIdentifier", Does Not Exist

although there is a bundle identifier.

eqyiel, 2017-05-05:
@itinance does this PR fix it for you? #692

skovhus, 2017-10-06:
Related: https://github.com/lwansbrough/react-native-camera/issues/827

LeBlaaanc, 2017-12-30:
I'm still getting
```
.clang.1_0.compiler
(1 failure)

Installing build/Build/Products/Debug-iphonesimulator/Example.app
An error was encountered processing the command (domain=NSPOSIXErrorDomain, code=2):
Failed to install the requested application
An application bundle was not found at the provided path.
Provide a valid path to the desired application bundle.
Print: Entry, ":CFBundleIdentifier", Does Not Exist

Command failed: /usr/libexec/PlistBuddy -c Print:CFBundleIdentifier build/Build/Products/Debug-iphonesimulator/Example.app/Info.plist
Print: Entry, ":CFBundleIdentifier", Does Not Exist

```

LeBlaaanc, 2017-12-30:
I get something even more obnoxious when trying to run it in Xcode. 
[image]

sibelius, 2017-12-30:
This does not work with expo, does anybody want to upgrade the example to latest React native version?

sibelius, 2018-01-23:
fixed on master

closed 2018-01-23 by sibelius