react-native-camera/react-native-camera #599: PDF417 cannot be read when it is parallel to phone screen

bosung90, 2017-02-14:
### Steps to reproduce
1. react-native init AppName
2. npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save
3. react-native link react-native-camera
4. Add onBarCodeRead and barCodeTypes to Camera as props
```
onBarCodeRead={(data) => {
                alert(data.data)
              }}
              barCodeTypes={['pdf417']}
```

### Expected behaviour
PDF417 should be read successfully regardless of rather phone is parallel or perpendicular to the barcode

### Actual behaviour
PDF417 barcode is only successfully read when the barcode is perpendicular to the phone.

### Environment
- **Node.js version**: 7.5.0
- **React Native version**: 0.41.2
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc
Android 6.0.1

### react-native-camera
**Version**: npm version or "master"
    "react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git"

semiautomatix, 2017-02-23:
I'm having a similar problem where the barcode isn't being read in landscape view - only portrait.

augusto-altman, 2018-06-02:
Facing the same issue here, any news on this?

SimonErm, 2018-11-21:
Does this issue still occure with RNCamera?

syadykin, 2018-12-13:
This might fix this issue: https://github.com/react-native-community/react-native-camera/pull/1994

sibelius, 2018-12-13:
test on master

closed 2018-12-13 by sibelius