react-native-camera/react-native-camera #604: [AVCaptureSession startRunning] startRunning may not be called between calls to beginConfiguration and commitConfiguration

kevinnguy, 2017-02-17:
Has anyone had this error before: `[AVCaptureSession startRunning] startRunning may not be called between calls to beginConfiguration and commitConfiguration`

This error typically appears when the app doesn't ask for camera permission but in my app, the user has already given camera access. This error comes up time to time and I haven't been able to figure out why it's happening. 

### Environment
- **Node.js version**: 7.5.0
- **React Native version**: 0.39.0
- **React Native platform + platform version**: iOS 10.2.1

### react-native-camera
**Version**: 0.4.1

flybayer, 2017-03-13:
I just got this crash as well! Did you figure anything out @kevinnguy?

**react-native** 0.42.0
**react-native-camera** 0.6.0

flybayer, 2017-03-13:
It looks like this line is the culprit: https://github.com/lwansbrough/react-native-camera/blob/master/ios/RCTCameraManager.m#L457

`startRunning` is being called from an async block which isn't safe in this context. I'm wondering why we are trying to manually restart the session here in case of an error? Maybe this block should just be replaced with an error message?
```objc
    [self setRuntimeErrorHandlingObserver:[NSNotificationCenter.defaultCenter addObserverForName:AVCaptureSessionRuntimeErrorNotification object:self.session queue:nil usingBlock:^(NSNotification *note) {
      RCTCameraManager *strongSelf = weakSelf;
      dispatch_async(strongSelf.sessionQueue, ^{
        // Manually restarting the session since it must have been stopped due to an error.
        [strongSelf.session startRunning];
      });
    }]];
```

flybayer, 2017-04-15:
I tried to fix this with a [this commit](https://github.com/flybayer/react-native-camera/commit/4f3a280cc5e80962b2696dfdd3a33c999d5b634e) in my fork. However, I'm still occasionally getting the crash, but maybe a little less often.

I really need this fixed! Can anyone offer any suggestions? I'm new to iOS & Obj-C and don't know what else to do.

/cc @lwansbrough @ptomasroos @NazarYablonskiy

flybayer, 2017-04-15:
So I think the crash wasn't happing where I thought it was at L457 like I said above. My crash logs using my attempted fix show the crash is still happening in `startSession`.

So I made [this commit](https://github.com/flybayer/react-native-camera/commit/1942882a93902a25c8aed77e108755933c2e0f76) in another attempt to fix the crash. I'll see how it goes.

Eyesonly88, 2017-04-19:
@flybayer can you confirm if that commit fixes it 🙏  ? I'm hitting this as well (`RCTCameraManager.m:453:5 __32-[RCTCameraManager startSession]_block_invoke`)

flybayer, 2017-04-20:
@Eyesonly88 I haven't pushed out the update with that commit yet, so I can't yet confirm. But I'm hoping to submit it for review this weekend.

Eyesonly88, 2017-04-20:
@flybayer Thank you SO MUCH 👍 🙇

flybayer, 2017-04-27:
@Eyesonly88 I'm still getting the crash 😞  

I'm seriously considering making a brand new react-native camera library with swift. Maybe build on a battle tested library build for stock iOS.

npomfret, 2017-04-27:
@flybayer A friend of mine wrote a camera app called NovaCamera for both iOS and android, it's excellent I have to admit - worth checking out.  Not as fully featured maybe (no bar code scanning, no video) but lots of stuff around exposure, focus and iso control.  But maybe we could learn from some of the stuff he did:

https://github.com/novaphotos/nova-ios-app/blob/develop/NovaCamera/Services/SSCaptureSessionManager.m#L280

https://github.com/novaphotos/nova-android-opencamera/blob/master/src/com/almalence/opencam/cameracontroller/CameraController.java#L3282

AidenMontgomery, 2017-05-30:
I am seeing this error every now and then. @flybayer did you ever manage to get rid of the crash?

flybayer, 2017-05-30:
No I haven't been able to fix it 😓

pvanliefland, 2017-06-07:
I got around this issue by subclassing the component and overriding the render method: basically replacing the camera element itself by a plain React Native <View> element when the camera permissions haven't been granted.

You can see the code here: https://gist.github.com/pvanliefland/8c2522a27daf4759b015e3345ba47dda

If the library author is interested, I can work on a pull request

flybayer, 2017-06-07:
@pvanliefland I think that's a different issue. The issue we are having occurs when all permissions have been enabled.

lukaleli, 2017-06-21:
anything new on that?

zachrnolan, 2017-07-03:
This bug is showing up occasionally for some of my users. Any updates on getting a PR created for this? @pvanliefland @lwansbrough

jsierles, 2017-07-13:
Also seeing this with permissions granted.

Cygnus2112, 2017-09-25:
Just ran into this. Anyone had any luck finding a workaround?

flybayer, 2017-09-26:
@Cygnus2112 no :( Someone needs to build react-native-non-crashing-camera

Cygnus2112, 2017-09-26:
Aargh. Okay, thanks for trying, @flybayer

AshokICreate, 2017-10-04:
```
2017-10-04 17:18:36.848474+0530 mobile[2587:224048] XPC connection interrupted
2017-10-04 17:18:36.856211+0530 mobile[2587:222045] *** Terminating app due to uncaught exception 'NSGenericException', reason: '*** -[AVCaptureSession startRunning] startRunning may not be called between calls to beginConfiguration and commitConfiguration'
*** First throw call stack:
(0x1828a7d38 0x181dbc528 0x18817780c 0x1005ab270 0x100e9d49c 0x100e9d45c 0x100eac110 0x100ea09a4 0x100ead104 0x100eb4100 0x1824d2fe0 0x1824d2c30)
libc++abi.dylib: terminating with uncaught exception of type NSException
(lldb) 
```

when i capture image i am getting this error.

[image]

phips28, 2017-10-09:
any solutions?

om2pkral, 2017-10-17:
Well, crash is quite self describing, doesn't it ? Sorry, I don't have time to do PR, leave it somebody who has. Serialising setCaptureQuality to sessionQueue fix this.

```
- (void)setCaptureQuality:(NSString *)quality
{
    #if !(TARGET_IPHONE_SIMULATOR)
        if (quality) {
            dispatch_async([self sessionQueue], ^{
                //DLog(@"setCaptureQuality: begin");
                [self.session beginConfiguration];
                if ([self.session canSetSessionPreset:quality]) {
                    self.session.sessionPreset = quality;
                }
                [self.session commitConfiguration];
                //DLog(@"setCaptureQuality: end");
            });
        }
    #endif
}
```

manask88, 2017-11-13:
hey @om2pkral I can create a PR to fix this based on your suggestion.  I don't have much experience with Native iOS development. Can you please explain why the changes you suggest will fix the problem?

manask88, 2017-11-14:
I've created a PR that fixes the problem using @om2pkral 's suggestion https://github.com/lwansbrough/react-native-camera/pull/980

om2pkral, 2017-11-14:
Hello,

it is basicaly due accesing camera session from multiple threads, 
wrapping it to sessionQueue serializes access and remove race condition.

Cheers,

PK

AshokICreate, 2017-11-14:
when @manask88 PR is going to merge. it's really important for us.

manask88, 2017-11-14:
I'm don't have write access to this repo @AshokICreate . You could help to get the attention of the maintainers in my PR https://github.com/lwansbrough/react-native-camera/pull/980

@om2pkral thanks for the explanation!

AshokICreate, 2017-11-14:
@lwansbrough, @kesha-antonov, @rt2zz, @ptomasroos, @corbt, @jtremback, @manask88  @om2pkral

closed 2018-09-14 by SimonErm