react-native-camera/react-native-camera #608: hasFlash always returns false on first call

wwwdata, 2017-02-20:
### Steps to reproduce
1. call the `hasFlash` instance method.
2. Just log the output, it will always be false on the first call. After starting the camera for the second time, it is true when there is a flash.

### Expected behaviour
Should return true for the first time as well. This already is a promise so I guess something is not correctly waiting for some initialization?

### Actual behaviour
Is false, when the camera starts for the second time it's true.

### Environment
- **Node.js version**: 7.50
- **React Native version**: 0.41
- **React Native platform + platform version**: iOS 10.2.1

### react-native-camera
**Version**: 0.6.0

molerat619, 2017-03-26:
Not sure if it has something to do with it, but when I start my app and try to use the flash light, it doesn't work either. The state (flashMode) comes from redux and at least the correct Icon is displayed. When I switch the mode to whatever (even back to the initial one), it works. So only the first time (until I change the mode again), it does not work.

wwwdata, 2017-03-26:
Yes that is another bug. There also is a Ticket open for it here.

Guillane, 2017-04-04:
In RCT_EXPORT_METHOD(hasFlash: ...) of RCTCameraManager.m, when replacing 
`AVCaptureDevice *device = [self.videoCaptureDeviceInput device];` with `AVCaptureDevice *device = [AVCaptureDevice defaultDeviceWithMediaType:AVMediaTypeVideo];`, it will work correctly for my use case (Back camera on iPads vs. iPhones, both taking video).

NordlingArt, 2017-11-25:
Will this ever be fixed?

liyyil, 2018-07-04:
and i noticed for Android devices for example tablets, hasFlash function seems not exist(_this.camera.hasFlash is not a function)

n1ru4l, 2018-11-20:
Please use RNCamera and open a new issue if you still experience this issue!

closed 2018-11-20 by n1ru4l