react-native-camera/react-native-camera #611: Barcode only scans at center of View

Maftalion, 2017-02-22:
### Steps to reproduce
1. Point camera to a code39 barcode.

### Expected behaviour
It should be able to read the barcode anywhere in the Camera's view.

### Actual behaviour
It only reads the barcode if exactly centered in the Camera view.

### Environment
- **Node.js version: 7.1**:
- **React Native version: 0.39.2,**:
- **React Native platform + platform version: iOS 10.2**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version: 0.4.1**: npm version or "master"

Maftalion, 2017-03-06:
nope, not sure why it's only scanning at the center and I wasn't able to find a workaround either.

E-P1983, 2017-03-14:
Same here. Help!

averypfeiffer, 2017-03-25:
Anyone have any luck? Is there anyway to manually trigger a scan? The "auto-detection" detects the same barcode like twenty different times

SimonErm, 2018-09-14:
This issue is closed because of inactivity.
Furthermore RCTCamera is deprecated now, use RNCamera instead. If this issue is also related to RNCamera, please open a new one.

closed 2018-09-14 by SimonErm

ntk0104, 2019-07-10:
Hi guys, any news?