react-native-camera/react-native-camera #612: White screen if camera type doesn't exist

TrekSoft, 2017-02-24:
### Steps to reproduce
1. Run the app on an emulator or device with no rear facing camera and type set to default back facing camera

### Expected behaviour
Should automatically switch to the front camera or a method should be provided to check if the back camera exists so that the proper camera can be set.

### Actual behaviour
A white screen shows and there's no way to tell if there is a back facing camera on the device until you call capture()

### Environment
Latest everything

### react-native-camera
**Version**: 0.60

beeglebug, 2017-05-23:
Is there any way to check if the device has a camera at all?

npomfret, 2017-05-24:
Not in this api yet, but you can do it natively (and would be a welcome PR!).  It wouldn't be required on iOS I guess.

closed 2018-09-14 by SimonErm