react-native-camera/react-native-camera #613: Barcode Scan Error

NajibRajak, 2017-02-25:
### Steps to reproduce
1. react-native init BarcodeProject
2. npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save
3. react-native link react-native-camera
4. add the following code to index.ios.js
/**
 * Sample React Native App
 * https://github.com/facebook/react-native
 * @flow
 */
import React, { Component } from 'react';
import {
  AppRegistry,
  Dimensions,
  StyleSheet,
  Text,
  TouchableHighlight,
  View
} from 'react-native';
import Camera from 'react-native-camera';

export default class BarcodeProject extends Component {
  scanBarcode(data) {
		alert('xxxx');
	}
  render() {
    return (
      <View style={styles.container}>
        <Camera
						ref={(cam) => {
						this.camera = cam;
						}}
						style={styles.preview}
						onBarCodeRead={code => this.scanBarcode(code)}
						barCodeTypes={[
							'ean13',
              'ean8',
							'code39',
              'upca',
              'upce',
              'code128'
						]}
						aspect={Camera.constants.Aspect.fill}>
					</Camera>
      </View>
    );
  }

  takePicture() {
    this.camera.capture()
      .then((data) => console.log(data))
      .catch(err => console.error(err));
  }
}

const styles = StyleSheet.create({
  container: {
    flex: 1
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center',
    height: Dimensions.get('window').height,
    width: Dimensions.get('window').width
  },
  capture: {
    flex: 0,
    backgroundColor: '#fff',
    borderRadius: 5,
    color: '#000',
    padding: 10,
    margin: 40
  }
});

AppRegistry.registerComponent('BarcodeProject', () => BarcodeProject);

### Expected behaviour
Tell us what should happen in xcode 
BarcodeProject[8553:1629805] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCaptureMetadataOutput setMetadataObjectTypes:] Unsupported type found - use -availableMetadataObjectTypes'
*** First throw call stack:
(0x1819091b8 0x18034055c 0x18914aa48 0x1001d5810 0x10047d258 0x10047d218 0x10048aaec 0x100480ce0 0x10048b088 0x10048ce2c 0x10048cb78 0x18099b2a0 0x18099ad8c)
libc++abi.dylib: terminating with uncaught exception of type NSException

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**: v7.2.1
     "react": "~15.4.0",
    "react-native": "0.41.2",
    "react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git"
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"
[image]

mariotacke, 2017-02-27:
Not specifying `barCodeTypes` prevents this from happening. Obviously it will still read all types of barcodes. Having the same issue right now.

chen236310, 2017-03-14:
I found this issue by write ‘barCodeTypes = {['qr']}',then I remove it ,the code can work .

hilkeheremans, 2017-03-21:
The `barCodeTypes` options are not identical across iOS and Android and are most likely the cause. For cross-platform QR code scanning, use:
```
import { Platform } from 'react-native'

// ...

        const barcodeTypes = Platform.select({
            ios: ['org.iso.QRCode'], // see https://github.com/lwansbrough/react-native-camera/issues/600 and http://stackoverflow.com/questions/22187919/constants-for-avcapturemetadataoutputs-metadataobjecttypes-property
            android: ['qr']
        })
```

hilkeheremans, 2017-03-21:
Also, for iOS, at least the following options are available:
```
"org.gs1.UPC-E",
"org.iso.Code39",
"org.iso.Code39Mod43",
"org.gs1.EAN-13",
"org.gs1.EAN-8",
"com.intermec.Code93",
"org.iso.Code128",
"org.iso.PDF417",
"org.iso.QRCode",
"org.iso.Aztec"
```

closed 2018-09-14 by SimonErm