react-native-camera/react-native-camera #616: question: clear exemple to store gps location in image

smolleyes, 2017-03-01:
hi

first, thanks for this module

i spend 2 hours trying everything but i can t  get the location stored in my images (android, testing from real phone)

no way to add a clear exemple to do this (if it s possible..) ?

thanks!

joshuapinter, 2017-03-12:
@smolleyes Would love to have this as well. Did you find out how to do it? Thanks!

npomfret, 2017-03-19:
hi - i've submitted a PR which I think fixes this on iOS (android to follow).

The way to use it is to something like:

  async _takePicture() {
      const options = {};
      if (this.state.position) {
        // position comes from: navigator.geolocation.watchPosition((position) => this.setState({position}))
        options.location = this.state.position;
      }
      const data = await this._camera.capture(options);
  }

Then when you use RN CameraRoll to retrieve the image it has the location info attached.  I'm looking for someone to help test my PR.  If you have any time it would be appreciated.

https://github.com/lwansbrough/react-native-camera/pull/635

koswarabilly, 2017-11-17:
@npomfret 
i have this code in my project

```
takePicture() {
    const options = {};
    options.location = this.state.location;
    this.camera.capture({ metadata: options })
      .then((data) => console.log(data))
      .catch(err => console.error(err));
  }
```
everytime i press capture the application stop working

closed 2018-09-14 by SimonErm