react-native-camera/react-native-camera #617: Wrong path returned by this.refs.camera.capture() when capture mode is cameraRoll (Android)

kev-slater-mhr, 2017-03-02:
Node version: 7.5
React-native-cli: 2.0.1
React-native: 0.41.2
Platform: Android 5.0
React-native-camera: 0.6.0

When taking a photo and the capture mode is cameraRoll, the path returned from this.refs.camera.capture() is the disk location of the image (i.e. file:///storage/emulated/0/DCIM/IMG_20170302_091011.jpg) as opposed to the images location in the camera roll (i.e. content://media/external/images/media/16329).

I suspect that the problem is down to the fact that no callback is provided in the call to MediaScannerConnection.scanFile in RCTCameraModule.java to capture (no pun intended!) the correct resource reference. 
This problem becomes more evident if you try and chain the camera view with a camera roll view and the image just taken does not immediately appear in the gallery of images (i.e. MediaScannerConnection has yet to complete its scanning).

npomfret, 2017-03-21:
Hi, this should be fixed in https://github.com/lwansbrough/react-native-camera/pull/637, would you like to test it?  

The currently implementation isn't wrong, its just there are 2 ways to represent the new image in android.  This PR adds a 2nd field to the response called 'mediaUri' which I believe is the thing you want.

kev-slater-mhr, 2017-03-21:
Hi, Will do hopefully within the next week. Thanks for the fix.

kev-slater-mhr, 2017-03-21:
Just given it a quick check and it looks good to me. Thanks for the resolution! Any ideas when this will be merged into master?

npomfret, 2017-03-21:
I hope later today.  Just testing out the other changes

npomfret, 2017-03-21:
If you could comment on the PR that would be helpful.

npomfret, 2017-03-27:
Can we close this?

closed 2018-02-22 by kev-slater-mhr