react-native-camera/react-native-camera #618: Wrong result video orientation when recording in landscape mode and laid down to 45 degrees

realjamie, 2017-03-02:
### Steps to reproduce
1. Lock the screen to portrait by using react-native-orientation (https://github.com/yamill/react-native-orientation): 
```
  componentDidMount() {
    Orientation.lockToPortrait();
  }
```

2. Setup the Camera as follows:
```
      <Camera
        ref="cam"
        style={styles.container}
        aspect={Camera.constants.Aspect.fit}
        captureAudio
        captureMode={Camera.constants.CaptureMode.video}
        captureTarget={Camera.constants.CaptureTarget.cameraRoll}
        keepAwake
        orientation={Camera.constants.Orientation.auto}
        type={Camera.constants.Type.front}>
          {this._renderMenu()}
      </Camera>
```
3. Rotate the device to landscape and lay it down to 45 degree.
3. Start recording for a while and stop it. Go to Photos to check the video.

### Expected behaviour
The video should be in landscape format.

### Actual behaviour
The video is in portrait format. (If the device is in landscape mode and perpendicular to the ground, the result video will be in landscape format, which is expected.)

### Environment
- **Node.js version**: v6.9.1
- **React Native version**: 0.42.0
- **React Native platform + platform version**: iOS 10.0.2
- ** react-native-orientation (https://github.com/yamill/react-native-orientation): "git+https://github.com/apprennet/react-native-orientation.git#task/compatibleWith0.40"

### react-native-camera
**Version**: 0.6.0

hugohow, 2017-03-29:
+1, have you find a solution for this ?

realjamie, 2017-03-29:
Not yet

hugohow, 2017-03-29:
Related to https://github.com/lwansbrough/react-native-camera/issues/581, I will try his solution (to keel mounted Camera component)

realjamie, 2017-03-29:
Thanks! I'll check it out.

RazerTang, 2017-07-25:
I encountered the same  problem  @realjamie  .I  still don't know how to fix this problem,can you show some code  how to fix this problem.

realjamie, 2017-07-25:
I actually haven't solved this issue

hugohow, 2017-07-25:
There is a solution in this issue https://github.com/lwansbrough/react-native-camera/issues/581. It's a hack and it demands a lot of changes in the code, but it works for me ;)

realjamie, 2017-07-25:
Thanks @hugohow

closed 2018-09-14 by SimonErm