react-native-camera/react-native-camera #621: onBarCodeRead "bounds"

semo94, 2017-03-05:
## Expected behaviour
onBarCodeRead event listener object should contains"bounds" property

### Actual behaviour
it's just contain "type" and "data" without "bounds"

### Environment
- **Node.js version*4.1.2*:
- **React Native version*0.42.0*:
- **React Native platform + platform version**: Android 6.0

### react-native-camera
**Version*0.6.0*: npm version or "master"

### I'm working on small augmented reality project and this issue will prevent me tracking QR on camera canvas, please help me !

marszall87, 2017-03-07:
I have the same problem, but running on Android 7.1.2 and Node 7.5.0 (same `react-native` and `react-native-camera`).

MocanuMarius, 2017-04-25:
@marszall87 @semo94  Hi , please look at my solution for this and the fork . https://github.com/lwansbrough/react-native-camera/issues/683

closed 2018-09-14 by SimonErm