react-native-camera/react-native-camera #622: An error was encountered processing the command (domain=NSPOSIXErrorDomain, code=2):

ratonjuancarlos, 2017-03-06:
### Steps to reproduce
1. cd Example/
2. npm install
3. Change IP for jsCodeLocation in AppDelegate.m file to localhost (`jsCodeLocation = 'localhost';`)

### Expected behaviour
Example should run

### Actual behaviour
Get this error message:

> The following build commands failed:
	CompileC /Users/pabloh/projects/react-native/cameraExample/Example/ios/build/Build/Intermediates/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/Objects-normal/x86_64/RCTCameraManager.o RCTCameraManager.m normal x86_64 objective-c com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)
Installing build/Build/Products/Debug-iphonesimulator/Example.app
An error was encountered processing the command (domain=NSPOSIXErrorDomain, code=2):
Failed to install the requested application
An application bundle was not found at the provided path.
Provide a valid path to the desired application bundle.
Print: Entry, ":CFBundleIdentifier", Does Not Exist
Command failed: /usr/libexec/PlistBuddy -c Print:CFBundleIdentifier build/Build/Products/Debug-iphonesimulator/Example.app/Info.plist
Print: Entry, ":CFBundleIdentifier", Does Not Exist

### Environment
- **Node.js version**: v7.4.0
- **React Native version**: 0.34.1
- **React Native platform + platform version**: iOS 9.0

### react-native-camera
**Version**: npm version or "master"

atharstar, 2018-02-14:
i am stuck at same issue

arapocket, 2018-03-06:
I got the same after updating to 1.0.1 from 0.13.0

sibelius, 2018-03-06:
upgrade RNCamera

arapocket, 2018-03-06:
@sibelius I updated and got the error. I don't think closing this is helping anyone.

In fact there is a much deeper issue at play here. I keep thinking I have found the source of the error but I'm going down a long rabbit hole it seems like.

nikitph, 2020-04-02:
any update on this

closed 2020-06-08 by stale[bot]