react-native-camera/react-native-camera #623: Question: How to use interleaved2of5, itf214, datamatrix?

Minishlink, 2017-03-06:
Hi, I would like to scan an interleaved2of5 barcode but I don't know how to get it working. In the [onBarCodeRead documentation](https://github.com/lwansbrough/react-native-camera#onbarcoderead), it states `interleaved2of (when available)`. What does this `when available` means? It's not really clear.

### Steps to reproduce
1. Add a onBarCodeRead listener with a console log of event data
2. Scan an interleaved2of5 barcode

### Expected behaviour
The event data should appear in the console.

### Actual behaviour
Nothing happens, the barcode is not read. (it works with other regular types)

### Environment
- **Node.js version**: 7.5.0
- **React Native version**: 0.41
- **React Native platform + platform version**: iOS 8+, Android 4.4+

### react-native-camera
**Version**: 0.6.0

Minishlink, 2017-04-07:
This issue is fixed on the HEAD of master :)

@ptomasroos, is it possible to publish a new release?

closed 2017-04-07 by Minishlink

vagnerSl, 2017-07-12:
+1 How to use only interleaved2of5 ?