react-native-camera/react-native-camera #626: stop failed: -1007

damly, 2017-03-08:
I set captureQuality={"720p"} and  get error.

E/MediaRecorder: stop failed: -1007
E/RCTCameraModule: Media recorder stop error.
                                                            java.lang.RuntimeException: stop failed.
                                                                at android.media.MediaRecorder.native_stop(Native Method)
                                                                at android.media.MediaRecorder.stop(MediaRecorder.java:1163)
                                                                at com.lwansbrough.RCTCamera.RCTCameraModule.releaseMediaRecorder(RCTCameraModule.java:400)
                                                                at com.lwansbrough.RCTCamera.RCTCameraModule.onError(RCTCameraModule.java:138)
                                                                at android.media.MediaRecorder$EventHandler.handleMessage(MediaRecorder.java:1082)
                                                                at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                at android.os.Looper.loop(Looper.java:150)
                                                                at android.app.ActivityThread.main(ActivityThread.java:5621)
                                                                at java.lang.reflect.Method.invoke(Native Method)
                                                                at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:794)
                                                                at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:684)

ketanbhardwaj, 2017-04-20:
@damly I am facing the same issue. Did you find any solution?

condorman, 2018-02-28:
@damly same error, on Android 7.0  and clean installation of react native and master version of react-native-camera. Any solution?

arapocket, 2018-05-23:
Same error. First I was getting 'Expected a Camera component' but after adding the mute option I got this.

watadarkstar, 2018-10-12:
I think this issue is due to re-rendering which causes the camera to re-render and stopRecording gets called. Do you call `setState` and if so where / when?

fxcosta, 2018-10-26:
This problem is definitely state. I had a similar problem today and all because of a component that was updated within RNCamera. The solution was to put it out and modify position via "position: absolute". The strangest thing is that before this error not happen. It appeared from nowhere.

closed 2019-04-18 by sibelius