react-native-camera/react-native-camera #629: Chunked recording?

GeoffreyPlitt, 2017-03-14:
We're on RN=0.42 and this component is working great for us!

We'd like to explore a more real-time use case, which would require chunks (say 5-20mb) of video to be captured, raising a new event (or firing a callback) with each new chunk, while the camera continues to record.

Any ideas/plans for this? Or anybody have any pointers for what would be involved, in case we try to do it ourselves?

bulats, 2018-05-16:
Any ideas how to implement this?

sibelius, 2018-05-16:
We can implement this using events

let's discuss this in react-native-community slack https://join.slack.com/t/rn-community/shared_invite/enQtMzY0MDgxODcwOTYyLWEwNWY3Njc0Mzg1MzllYWVhNTU4NTY5MGY5NjQ2ZWM1MGFkY2QzMTljYWY2NzE0NTU4MmM3NTBkY2ZiZmIzMWY

#react-native-camera channel

sibelius, 2018-05-16:
you gonna need a new event here
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/CameraViewManager.java#L16

on android side
and emit some events like this
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java#L180

closed 2019-06-08 by stale[bot]