react-native-camera/react-native-camera #630: Android device of System version 7.0 can't run react-native-camera

yangzhenxing001, 2017-03-15:
### Steps to reproduce
1.- **Node.js version**:v7.3.0
- **React Native version**:0.39.2
 Android device of System version 5.0 or 6.0 run OK,but Android device of System version 7.0 can't run react-native-camera. Cause Application flash back。

I want to have a solution
 

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**:v7.3.0
- **React Native version**:0.39.2
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

npomfret, 2017-03-27:
Do you have a stack trace from the crash?

syardumi, 2017-03-31:
I have the same problem on mine. I don't get an error as I expected when the component tries to mount, but I do get one when I try to invoke the capture() method. 

`ReactNativeJS: { [Error: No camera found.] framesToPop: 1, code: 'EUNSPECIFIED' }`

syardumi, 2017-04-03:
I get the same error on my Nexus 5x running 6.0.1 also. Am I missing something?

acharlop, 2018-01-09:
Similar error here

**ENV**
- node: 8.8.1
- react-native: 0.49.5
- device: Pixel 2 XL
- android: 8.1.0
- react-native-camera: 0.12.0

**Error**
`{ "framesToPop": 1, "code": "EUNSPECIFIED" }`

**Steps taken**
1. Added a debugger instead of just printing the error
```
this.camera.capture()
    .then(...)
    .catch(e => {
        console.tron.log(e)
        debugger
    })
```
2. Debugger showed the full error message
`err = Error: /storage/emulated/0/Pictures/IMG_20180109_122934.jpg (Permission denied)`
3. In settings I toggled the storage permissions off and back on and that fixed it.

This is a **Short term solution** that worked for me not sure if that will work for you or even if that's the underlying bug. I see there's a branch called `android-6-permissions` perhaps this is what is being addressed there.

ospfranco, 2018-01-09:
Permission handling for Android 6+ is implemented on master and should be available in next release.

closed 2018-01-09 by ospfranco