react-native-camera/react-native-camera #632: help with buddybuild android crash

adrai, 2017-03-17:
I don't know exactly how this happens. Can someone help me in spotting where the issue could be?

=> https://github.com/lwansbrough/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraViewFinder.java#L284
```
java.lang.RuntimeException: getParameters failed (empty parameters)

FRAME	MODULE	CRASH DETAILS
0	android​.hardware	
Camera.native_getParameters
1	android​.hardware	
Camera.java ​ Line 1996
Camera.getParameters
2	com​.lwansbrough​.RCTCamera	
RCTCameraViewFinder.java ​ Line 284
RCTCameraViewFinder$ReaderAsyncTask.doInBackground
3	com​.lwansbrough​.RCTCamera	
RCTCameraViewFinder.java ​ Line 269
RCTCameraViewFinder$ReaderAsyncTask.doInBackground
4	android​.os	
AsyncTask.java ​ Line 295
AsyncTask$2.call
5	java​.util​.concurrent	
FutureTask.java ​ Line 237
FutureTask.run
6	android​.os	
AsyncTask.java ​ Line 234
AsyncTask$SerialExecutor$1.run
7	java​.util​.concurrent	
ThreadPoolExecutor.java ​ Line 1113
ThreadPoolExecutor.runWorker
8	java​.util​.concurrent	
ThreadPoolExecutor.java ​ Line 588
ThreadPoolExecutor$Worker.run
9	java​.lang	
Thread.java ​ Line 818
Thread.run
```

my render function returns something like this:
```jsx
      <View onLayout={this.onLayout} style={styles.camera}>
        <Camera type={cameraSide || 'front'} onBarCodeRead={this.onBarCodeRead} style={styles.camera}>
          <View style={[styles.rectangleContainer, found ? styles.rectangleContainerFound : null]}>
            <Animated.View style={[styles.rectangle, found ? styles.rectangleFound : null, { width: rHA, height: rHA }]}>
              <View style={found ? null : [styles.rectangleInner, { width: riH, height: riH, top: 0, left: 0, borderTopWidth: 2, borderLeftWidth: 2 }]} />
              <View style={found ? null : [styles.rectangleInner, { width: riH, height: riH, top: 0, right: 0, borderTopWidth: 2, borderRightWidth: 2 }]} />
              <View style={found ? null : [styles.rectangleInner, { width: riH, height: riH, bottom: 0, left: 0, borderBottomWidth: 2, borderLeftWidth: 2 }]} />
              <View style={found ? null : [styles.rectangleInner, { width: riH, height: riH, bottom: 0, right: 0, borderBottomWidth: 2, borderRightWidth: 2 }]} />
              {
                found && <Icon name="icon-success" success xlarge />
              }
            </Animated.View>
          </View>
        </Camera>
      </View>
```

### Steps to reproduce
1. Go to a view that renders the Camera component
2. Switch to an other view that renders the Camera component

### Expected behaviour
No crash ;-)

### Actual behaviour
App crashes

### Environment
- **Node.js version**: v6.9.10
- **React Native version**: v0.43.0-rc.2
- **React Native platform + platform version**: Android 6.0.1

### react-native-camera
**Version**: 0.6.0

adrai, 2017-03-22:
Seems to be this issue: https://github.com/lwansbrough/react-native-camera/issues/591

adrai, 2017-03-23:
More info:

```
java.lang.RuntimeException: Camera is being used after Camera.release() was called

FRAME	MODULE	CRASH DETAILS
0	android​.hardware	
Camera.setHasPreviewCallback
1	android​.hardware	
Camera.java ​ Line 152
Camera.access$600
2	android​.hardware	
Camera.java ​ Line 1181
Camera$EventHandler.handleMessage
3	android​.os	
Handler.java ​ Line 102
Handler.dispatchMessage
4	android​.os	
Looper.java ​ Line 158
Looper.loop
5	android​.app	
ActivityThread.java ​ Line 7229
ActivityThread.main
6	java​.lang​.reflect	
Method.invoke
7	com​.android​.internal​.os	
ZygoteInit.java ​ Line 1230
ZygoteInit$MethodAndArgsCaller.run
8	com​.android​.internal​.os	
ZygoteInit.java ​ Line 1120
ZygoteInit.main
```

ThaJay, 2018-01-16:
Thank you for reposting this issue as the original was closed. I will repost my log here.
https://github.com/react-native-community/react-native-camera/issues/357
https://github.com/react-native-community/react-native-camera/issues/357#issuecomment-353083409
1. navigate to camera screen
2. scan barcode - display modal
3. press back button
4. crash

This does not always happen, it can also navigate back like it's supposed to, but it does have a pretty high chance of happening.

"react-native": "^0.50.4"
"react-native-camera": "^0.12.0"
"react-navigation": "^1.0.0-beta.22"

```
E/AndroidRuntime(13470): FATAL EXCEPTION: AsyncTask #1
E/AndroidRuntime(13470): Process: com.chipta, PID: 13470
E/AndroidRuntime(13470): java.lang.RuntimeException: An error occured while executing doInBackground()
E/AndroidRuntime(13470): 	at android.os.AsyncTask$3.done(AsyncTask.java:300)
E/AndroidRuntime(13470): 	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:355)
E/AndroidRuntime(13470): 	at java.util.concurrent.FutureTask.setException(FutureTask.java:222)
E/AndroidRuntime(13470): 	at java.util.concurrent.FutureTask.run(FutureTask.java:242)
E/AndroidRuntime(13470): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:231)
E/AndroidRuntime(13470): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
E/AndroidRuntime(13470): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
E/AndroidRuntime(13470): 	at java.lang.Thread.run(Thread.java:841)
E/AndroidRuntime(13470): Caused by: java.lang.RuntimeException: Method called after release()
E/AndroidRuntime(13470): 	at android.hardware.Camera.native_getParameters(Native Method)
E/AndroidRuntime(13470): 	at android.hardware.Camera.getParameters(Camera.java:1765)
E/AndroidRuntime(13470): 	at com.lwansbrough.RCTCamera.RCTCameraViewFinder$ReaderAsyncTask.doInBackground(RCTCameraViewFinder.java:309)
E/AndroidRuntime(13470): 	at com.lwansbrough.RCTCamera.RCTCameraViewFinder$ReaderAsyncTask.doInBackground(RCTCameraViewFinder.java:294)
E/AndroidRuntime(13470): 	at android.os.AsyncTask$2.call(AsyncTask.java:288)
E/AndroidRuntime(13470): 	at java.util.concurrent.FutureTask.run(FutureTask.java:237)
E/AndroidRuntime(13470): 	... 4 more
```

redbluenat, 2018-03-06:
any progess with this issue?

sibelius, 2018-03-06:
Upgrade to RNCamera

jqn, 2018-06-26:
Same problem exists in RNCamera

sibelius, 2018-06-26:
@jqn open another issue with more information

shahchaitanya, 2018-08-02:
Same Problem with Rn Camera. It crash when we use back button in android.

devapro, 2018-08-07:
@shahchaitanya try to use latest code from the master branch. I used from master in production - work well.

ThaJay, 2018-08-09:
@shahchaitanya I have not had complaints about this crash in some time. I also did some things in the meantime to prevent double navigation at some point because that also resulted in other issues.
I think this bug has something to do with double rendering of the camera but I can't be sure because I was never able to confirm with a good test.

All I know is that our app has not been crashing much lately and when it did it was not because of this.

SimonErm, 2018-09-25:
Can this be closed?

n1ru4l, 2018-12-19:
Closed due to missing activity. If you still experience this issue with the newest version of `RNCamera` please open a new issue.

closed 2018-12-19 by n1ru4l