react-native-camera/react-native-camera #638: App crashed after taking photo on Android

minhchu, 2017-03-21:
### Actual behaviour
On Samsung Galaxy J7 Android 6.0, app crashed after taking a photo.
Here is the stacktrace I got from the Google Play:
```
java.lang.OutOfMemoryError: Failed to allocate a 51121164 byte allocation with 16777216 free bytes and 36MB until OOM
	at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
	at android.graphics.Bitmap.nativeCreate(Native Method)
	at android.graphics.Bitmap.createBitmap(Bitmap.java:975)
	at android.graphics.Bitmap.createBitmap(Bitmap.java:946)
	at android.graphics.Bitmap.createBitmap(Bitmap.java:877)
	at com.lwansbrough.RCTCamera.RCTCameraModule.rotate(RCTCameraModule.java:562)
	at com.lwansbrough.RCTCamera.RCTCameraModule.fixOrientation(RCTCameraModule.java:581)
	at com.lwansbrough.RCTCamera.RCTCameraModule.access$300(RCTCameraModule.java:45)
	at com.lwansbrough.RCTCamera.RCTCameraModule$3.onPictureTaken(RCTCameraModule.java:673)
	at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1165)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:158)
	at android.app.ActivityThread.main(ActivityThread.java:7230)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
```

### Environment
- **Node.js version**: 6.8.1
- **React Native version**: 0.41.0
- **React Native platform + platform version**: Android 7

### react-native-camera
**Version**: 0.6.0

adrai, 2017-03-22:
Seems to be this issue: https://github.com/lwansbrough/react-native-camera/issues/591

npomfret, 2017-03-27:
@minhchu can you try the master branch?  there are some changes in there that have not been released to npm yet which might help.

skovhus, 2017-04-20:
I think this is a duplicate of #590.

The problem is the rotation done after the image is taken (introduced in 0.4.1). On some devices there is not enough memory for doing this operation... We should investigate why this rotation is even necessary. 

One solution for you might be to use react-native-camera@0.4.0 (if you are using React Native < 0.40.0).

npomfret, 2017-04-24:
@skovhus are you able to test this: https://github.com/lwansbrough/react-native-camera/pull/682

it removes image rotation but i can't properly test it on android because the device i'm using rotate the image for me.

closed 2018-09-14 by SimonErm