react-native-camera/react-native-camera #642: Using Two Camera Components in an Application Causing Weird Behavior?

MaxwellGover, 2017-03-23:
I'm trying to create two different camera components in my application. One is to take a picture of the front of a product and save the path to my redux store under `frontImage`. The second camera component is to take a picture of the back of a product and save the image path to redux store under `backImage`. 

The second camera component is overwriting the `frontImage` path. This is not because I was dispatching the wrong redux action. I comment out everything inside of the `takePicture()` function inside of the second component and it turns out that `onPress()` it is running the `takePicture()` function inside of the first camera component instead. 

I have no idea why this is. Not even sure how the second camera component would even know about the first camera component? 

### Steps to reproduce
1. Create two different camera components. Ex. CameraOne & CameraTwo in separate files.
2. Try `console.log('Picture from Camera Two')` inside of CameraTwo component and `console.log('Picture from Camera One')` inside of CameraOne component.
3. Click the capture button inside of the CameraTwo component to run the `takePicture()` method.

### Expected behaviour
It seems like when running the `takePicture()` function from within CameraTwo component you would get 'Picture from Camera Two'.

### Actual behaviour
CameraTwo component is running the `takePicture()` function from CameraOne component instead thus logging 'Picture from CameraOne'

### Environment
- **React Native** 0.42

### react-native-camera
Latest

npomfret, 2017-03-27:
I've also had problems trying to make 2 camera components.  I'm just not sure it's supported.

MihaiDamian, 2017-04-19:
I had a look over the native iOS implementation. In the current form the library definitely does not support multiple cameras.

If you look at RCTCameraManager you will see that a view is created and cached until the camera view is removed from the DOM. If you try to present a second camera the same view (RCTCamera) will be used, together with a bunch of objects and properties that are kept in RCTCameraManager.

My advice is to rethink your app flows and try to avoid multiple cameras. If you're absolutely sure you still need multiple cameras, have a look at my [fork](https://github.com/MihaiDamian/react-native-camera). 

The basic idea is you have to return a new camera view object from RCTCameraManager whenever React Native asks for it and you can't keep any property in RCTCameraManager  that's specific to a single camera (that includes the camera view itself). 

The other problem that you'll run into is the library needs to set properties on AVCaptureDevice objects (for things like flash settings). These objects are created by the system and you guessed it, they are one per physical device (and thus shared between all camera instances). I don't have a simple solution for handling them from within the library. One way to do it is to keep these camera properties in your component's state and re-render the component when it's navigated back to.

I didn't look over the Android code but I understand there are similar problems there as well.

MaxwellGover, 2017-04-19:
@MihaiDamian Thanks! What I ended up doing was creating a `frontImageTaken` and `backImageTaken` properties in my Redux store (you might be able to use local state too, I haven't tried) then just used `if/else` statements to get the functionality I needed. So if an image of the front of the product was already taken that would mean a photo of the back of the product still needs to be taken and the correct methods would then be called.

jrojers, 2017-08-28:
The @MihaiDamian fork works for me too. Fixes issue logged here: https://github.com/lwansbrough/react-native-camera/issues/721.

closed 2017-09-21 by MaxwellGover

bennybennet, 2017-09-22:
@MaxwellGover 
So to fix this we are supposed to use the fork or is it pulled into the original?

unknouun, 2021-02-07:
Has this issue been addressed with the release of AVCaptureMultiCamSession?