react-native-camera/react-native-camera #648: How can I take a square photo?

cserdean, 2017-03-25:
Sorry for asking here, I know this isn't an issue per say, but I need to know how I can take square photos. I managed to get the preview to be squared but the photos taken where still rectangular.

xilonxiao, 2017-05-09:
+1
I don't if it applies here, but I remember once I wanted to accomplish this same effect when implementing an native android application. I had to crop the photo manually. Maybe it is also de case here.

RekkyRek, 2017-07-06:
You can use [imagemagick](https://www.npmjs.com/package/imagemagick) to crop it after capture.

bhrzk, 2017-09-09:
How about square video like instagram ? ( 720x720 for example )

sibelius, 2018-01-07:
I think this is better on stackoverflow

you can always take a photo and crop it later on

check this https://stackoverflow.com/questions/42331823/crop-image-with-react-native

closed 2018-01-07 by sibelius

ahmetilhann, 2018-06-04:
I find solution with `ImageEditor`
```
import React, { Component } from 'react';
import { Text, View, TouchableOpacity, Dimensions, StyleSheet, ImageEditor, ToastAndroid } from 'react-native';
import { RNCamera } from 'react-native-camera';

const { width } = Dimensions.get('window');

class CameraIndexScene extends Component {
    constructor(props) {
        super(props);

        this.state = {
            flashMode: RNCamera.Constants.FlashMode.off,
            type: RNCamera.Constants.Type.back,
            loading: false,
        };

        this.takePicture = this.takePicture.bind(this);
    }
    takePicture() {
        ToastAndroid.show('Please waiting...', ToastAndroid.LONG);
        if (this.camera) {
            this.setState({ loading: true });
            const options = { quality: 1, base64: false };
            this.camera.takePictureAsync(options)
                .then((data) => {
                    const cropData = {
                        offset: {
                            x: 0,
                            y: 0,
                        },
                        size: {
                            width: data.width, 
                            height: data.width,
                        },
                    };
                    ImageEditor.cropImage(
                        data.uri, 
                        cropData, 
                        (successURI) => { 
                            console.log(successURI);
                        }, 
                        (error) => {
                            console.log(error.message);
                        },
                    );
                });
        }
    }

    render() {
        return (
            <View style={styles.container}>
                <RNCamera
                    ref={(ref) => {
                        this.camera = ref;
                    }}
                    style={styles.preview}
                    type={this.state.type}
                    flashMode={this.state.flashMode}
                    permissionDialogTitle="Permission to use camera"
                    permissionDialogMessage="We need your permission to use your camera phone"
                    ratio="1:1"
                />
                <View style={styles.buttonContainer}>
                    <TouchableOpacity 
                        style={styles.cameraButton}
                        onPress={this.takePicture}
                        disabled={this.state.loading}
                    >
                        <Text>GO</Text>
                    </TouchableOpacity>
                </View>
            </View>
        );
    }
}

const styles = StyleSheet.create({
    container: {
        flex: 1,
        flexDirection: 'column',
        backgroundColor: 'black',
    },
    preview: {
        alignItems: 'center',
        width,
        height: width,
    },
    buttonContainer: {
        flex: 2, 
        flexDirection: 'row', 
        backgroundColor: 'white',
        justifyContent: 'center',
        alignItems: 'center',
    },
    cameraButton: {
        borderWidth: 2,
        borderColor: 'red',
        height: 90,
        width: 90,
        margin: 20,
        justifyContent: 'center',
        alignItems: 'center',
        borderRadius: 45,
    },
});

export default CameraIndexScene;
```

eqyiel, 2018-06-15:
https://github.com/react-native-community/react-native-camera/pull/1636

redaben, 2018-07-01:
@bhrzk did you find any solutions for square video like instagram ?

muratozgul, 2018-10-09:
setting ratio="1:1" on RNCamera worked for me on Android

```
<RNCamera
  ref={ref => this.cameraRef = ref}
  style={styles.cameraPreview}
  type={this.state.cameraType}
  ratio='1:1'
/>
```

Shahanshah87, 2022-01-05:
> ratio='1:1'

> setting ratio="1:1" on RNCamera worked for me on Android
> 
> ```
> <RNCamera
>   ref={ref => this.cameraRef = ref}
>   style={styles.cameraPreview}
>   type={this.state.cameraType}
>   ratio='1:1'
> />
> ```

Hey can you please share the whole code of Camera because `ratio="1:1"` is not working for me.