react-native-camera/react-native-camera #649: Exception thrown on failed to focus

alialamine, 2017-03-25:
### Steps to reproduce
1. Find a phone that has a really blurry camera
2. Try to focus with it

### Expected behaviour
return a handler, or at least not crash because of these few devices

### Actual behaviour
The app crashes and exits. Here is the exception:
```
Fatal Exception: java.lang.RuntimeException: autoFocus failed
       at android.hardware.Camera.native_autoFocus(Camera.java)
       at android.hardware.Camera.autoFocus(Camera.java:975)
       at com.lwansbrough.RCTCamera.RCTCameraViewFinder.handleFocus(RCTCameraViewFinder.java:374)
       at com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent(RCTCameraViewFinder.java:341)
       at android.view.View.dispatchTouchEvent(View.java:7565)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2211)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1954)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2211)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1954)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2211)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1954)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2211)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1954)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2211)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1954)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2211)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1954)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2211)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1954)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2211)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1954)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2211)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1954)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2211)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1954)
       at com.android.internal.policy.impl.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:1973)
       at com.android.internal.policy.impl.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1425)
       at android.app.Activity.dispatchTouchEvent(Activity.java:2424)
       at com.android.internal.policy.impl.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:1921)
       at android.view.View.dispatchPointerEvent(View.java:7745)
       at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:3909)
       at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:3804)
       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3399)
       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3449)
       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3418)
       at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:3503)
       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3426)
       at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:3560)
       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3399)
       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3449)
       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3418)
       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3426)
       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3399)
       at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:5446)
       at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:5426)
       at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:5397)
       at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:5520)
       at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:182)
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:136)
       at android.os.Looper.loop(Looper.java:197)
       at android.app.ActivityThread.main(ActivityThread.java:5225)
       at java.lang.reflect.Method.invokeNative(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:525)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:741)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:557)
       at dalvik.system.NativeStart.main(NativeStart.java)
```

### Environment
- **Node.js version**: 7.7.1
- **ReactNative**: 0.42.0
-  **OS**: Android 4.4

### react-native-camera
**Version**: 0.6.0

iegik, 2017-06-22:
It also happens then camera is not available because battery is low. So, component is launched, I try to focus by pressing on camera view - then app crashed.

Версия Android
Android 6.0
Устройство
ZTE Blade A210 (ZTE_Blade_A210)
Производитель ZTE ОЗУ (МБ) 1024 Размер экрана 480 × 854 Разрешение экрана (dpi) 240 Аппаратная платформа armeabi-v7a Версия OpenGL ES 3.0 Марка ЦП Mediatek Модель ЦП MT6735M

```
java.lang.NullPointerException: 
  at com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent(RCTCameraViewFinder.java:326)
  at android.view.View.dispatchTouchEvent(View.java:9415)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2257)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2257)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2257)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2257)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2257)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2257)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2257)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2257)
  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2257)
  at com.android.internal.policy.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2453)
  at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1772)
  at android.app.Activity.dispatchTouchEvent(Activity.java:2808)
  at com.android.internal.policy.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2414)
  at android.view.View.dispatchPointerEvent(View.java:9646)
  at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4743)
  at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4601)
  at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4106)
  at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4159)
  at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4125)
  at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4271)
  at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4133)
  at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:4328)
  at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4106)
  at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4159)
  at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4125)
  at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4133)
  at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4106)
  at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:6560)
  at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:6534)
  at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:6477)
  at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6732)
  at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
  at android.os.MessageQueue.nativePollOnce(MessageQueue.java:0)
  at android.os.MessageQueue.next(MessageQueue.java:328)
  at android.os.Looper.loop(Looper.java:164)
  at android.app.ActivityThread.main(ActivityThread.java:5728)
  at java.lang.reflect.Method.invoke(Method.java:0)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:789)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:679)
```

iegik, 2017-08-28:
https://stackoverflow.com/questions/10100351/android-fail-to-connect-to-camera-service

closed 2018-09-14 by SimonErm