react-native-camera/react-native-camera #650: Capture not working on Android 4.4

molerat619, 2017-03-25:
### Steps to reproduce
Using this redux and container, I can't capture images on my Android 4.4 device. It works in the emulator.

**Camera Component**
```
import React, { Component, PropTypes } from 'react'
import { ScrollView, View, Text, Image, TouchableOpacity, KeyboardAvoidingView } from 'react-native'
import { connect } from 'react-redux'
import Camera from 'react-native-camera'
import CameraActions from '../Redux/CameraRedux'

// Add Actions - replace 'Your' with whatever your reducer is called :)
// import YourActions from '../Redux/YourRedux'
import { Metrics } from '../Themes'

// external libs
import Icon from 'react-native-vector-icons/FontAwesome'
import Animatable from 'react-native-animatable'
import { Actions as NavigationActions } from 'react-native-router-flux'

// Styles
import styles from './Styles/CameraScreenStyles'

// I18n
import I18n from 'react-native-i18n'

class CameraScreen extends Component {
  static propTypes = {
    dispatch: PropTypes.func,
    lastPic: PropTypes.string,
    capture: PropTypes.func
  }

  constructor (props) {
    super(props)
  }

  takePicture() {
    const options = {};
    //options.location = ...
    this.camera.capture({metadata: options})
      .then((data) => {
        // console.log(data)

        this.props.capture(data)
      })
      .catch(err => console.error(err));
  }

  render () {
    return (
      <View style={styles.container}>
        <Camera ref={(cam) => { this.camera = cam; }} style={styles.preview} aspect='stretch'>
          <View style={styles.toolsContainer}>
            <Image source={{uri: this.props.lastPic, isStatic:true}} style={ styles.tool, styles.lastPic } />

            <Text style={{ color: '#fff' }}>
              { this.props.lastPic }
            </Text>
            <TouchableOpacity onPress={this.takePicture.bind(this)}>
              <Text style={ styles.tool, styles.capture }>
                Capture
              </Text>
            </TouchableOpacity>
          </View>
        </Camera>
      </View>
    )
  }

}

const mapStateToProps = (state) => {
  return {
    lastPic: state.camera.lastPic
  }
}

const mapDispatchToProps = (dispatch) => {
  return {
    capture: (data) => dispatch(CameraActions.capture(data))
  }
}

export default connect(mapStateToProps, mapDispatchToProps)(CameraScreen)
```

**Camera Redux**
```
import { createReducer, createActions } from 'reduxsauce'
import Immutable from 'seamless-immutable'

/* ------------- Types and Action Creators ------------- */

const { Types, Creators } = createActions({
  lastPic: null,
  capture: ['data']
})

export const CameraTypes = Types
export default Creators

/* ------------- Initial State ------------- */

export const INITIAL_STATE = Immutable({
  lastPic: null
})

/* ------------- Reducers ------------- */

// save the captured image as lastPic so it can be used and displayed
export const capture = (state, { data }) => {
  return state.merge({ lastPic: data.path });
}

/* ------------- Hookup Reducers To Types ------------- */

export const reducer = createReducer(INITIAL_STATE, {
  [Types.CAPTURE]: capture
})
```

### Emulator vs. Real Device behaviour
When I click the capture button in the emulator, it captures an image and displays it in the <Image> component. It is updated with every capture, just like the text displaying the path. On my phone, nothing happens when I click the button.

### Environment
- **Node.js version**:  7.7.1
- **React Native version**: 4.2.1
- **React Native platform + platform version**: Android 4.4

### react-native-camera
**Version**: "master"

KatarinaSimnova, 2017-03-27:
Same problem on Android 4.1.2.

molerat619, 2017-04-05:
It's working on my Android 4.4 device now, but I have no idea why. That means it was most probably not an issue with the module but with my code. If I find out what the problem was, I will share it.

closed 2017-04-05 by molerat619

mike-aungsan, 2017-07-12:
I am having the same issue with Android 4.4 S4. Any help?