react-native-camera/react-native-camera #652: barCodeTypes error on iOS

letops, 2017-03-30:
### Expected behaviour
We should only have to set an array with the barcode types that we wish to detect, as the documentation explains: `[ 'aztec', 'code128', ...etc ]`

### Actual behaviour
This is not respected by iOS. In iOS you need to set the barcode types in a different manner, e.g. `[ 'org.iso.Aztec', 'org.iso.Code128', ...etc ]`, if you don't the app freezes and crashes.

### Environment
- **React Native version**: 0.41.2
- **React Native platform + platform version**: iOS 10.3

### Some other findings
On ANDROID only:
```javascript
    barCodeTypes = {[
      'aztec',
      'codabar',  // Not in the docs, but found in code
      'code128',
      'code39',
      // 'code39mod43',  // Found in the docs but not in the code
      'code93',
      'datamatrix',  // When available(?), according to the docs
      'ean13',
      'ean8',
      'interleaved2of5',  // When available(?), according to the docs
      // 'itf14',  // When available(?), according to the docs, but not found in code
      'maxicode',  // Not in the docs, but found in code
      'pdf417',
      'qr',
      'rss14',  // Not in the docs, but found in code
      'rssexpanded',  // Not in the docs, but found in code
      'upca',
      'upce',
      'upceanextension',
    ]
```

On IOS only: (#613 is the only place where this is commented)

```javascript
    barCodeTypes = {[
        'org.iso.Aztec',
        'org.iso.Code128',
        // 'code138' should be here, but not sure of the namespace (found in code)
        'org.iso.Code39',
        'org.iso.Code39Mod43',
        'com.intermec.Code93',
        // 'datamatrix' should be here, but not sure of the namespace (found in code)
        'org.gs1.EAN-13',
        'org.gs1.EAN-8',
        // 'interleaved2of5' should be here, but not sure of the namespace (found in code)
        // 'itf14' should be here, but not sure of the namespace (found in code)
        'org.iso.PDF417',
        'org.iso.QRCode',
        'org.gs1.UPC-E',
    ]}
```

benvium, 2017-11-03:
On iOS, the string to use for datamatrix is `org.iso.DataMatrix` (available since iOS8 https://developer.apple.com/documentation/avfoundation/avmetadataobjecttypedatamatrixcode)

niksajanjic, 2018-02-02:
Per documentation, this issue should've been resolved by using `Camera.constants.barCodeType`, but it doesn't work for me. See: https://github.com/react-native-community/react-native-camera/issues/1206

n1ru4l, 2018-12-19:
Please test it with the newest version of RNCamera and open a new issue if this error still occurs.

closed 2018-12-19 by n1ru4l