react-native-camera/react-native-camera #655: delay to open camera on android

ronaldozanoni, 2017-04-03:
The camera is delaying to open and to take the photo on android devices. In ios, is working great.

Take a look in the screen recorded below.

### Steps to reproduce
1. Clone the repo: https://github.com/ronaldozanoni/react-native-camera-test-android
2. Click on "New" on header
3. Check the 1 second delay to open

### Environment
- ** react-native-cli: 2.0.1
- **Node.js version**: 7.0.0
- **React Native version**: 0.42.3

Tested on Motorola Moto X 1, Android 5.1

Am I doing anything wrong? 

Thanks.

Ronaldo Zanoni

[image]

molerat619, 2017-04-05:
Similar with my Android 4.4 device.

OmriAharon, 2017-04-28:
Delay in opening camera for me as well

itxing666, 2017-06-02:
Delay in opening camera for me as well, i want to have a solution

efnan34, 2017-06-12:
When i click to navigate camera it makes appx. 500 ms delay !

Symous, 2017-08-08:
Hi,guys,I have found this issue before and modified the android native code to solve it by adding a prop called "onRecordVideoStarted" ,so that we can know when does the camera is ready for work. I make a PR but the author haven't merge it yet. see it here : #703  :)

closed 2018-09-14 by SimonErm