react-native-camera/react-native-camera #659: Flash causes App to crash on Android 4.4

molerat619, 2017-04-05:
### Steps to reproduce
1. Take a picture with flash mode on (using Android 4.4 device)
2. App will crash

### Expected behaviour
Should start flash light and take a picture.

### Actual behaviour
App crashes.

### Environment
- **Node.js version**: 7.x
- **React Native version**: 0.42
- **React Native platform + platform version**: Android 4.4

### react-native-camera
**Version**: "master"

skovhus, 2017-04-20:
@molerat619 any stack trace?

molerat619, 2017-04-20:
@skovhus how can I get that when running on my phone? Sorry, I'm quite new to react native.

skovhus, 2017-04-20:
When your phone is connected to your computer, you can run (unix or OS X):

`adb logcat '*:S ReactNative:V ReactNativeJS:V RCTCamera:V'`

On Windows, I guess you need to open Android Studio.

molerat619, 2017-04-20:
Thanks for your help. Unfortunately, after doing so, I still get an empty log:
[image]

skovhus, 2017-04-20:
Ah yeah, I guess you need to filter by your apps name. 

`adb logcat '*:S myappname:V'` 

If this doesn't work you can always try:

`adb logcat '*:E'` (all errors)
`adb logcat '*:F'` (all fatals)

molerat619, 2017-04-20:
I guess it's this one then:

[image]

skovhus, 2017-04-20:
Is this only happening when using the Flash? The crash happens when the library tries to fix orientation after the image is taken. So related to #590 #638.

If you are using RN < 0.40 you can use react-native-camera@0.4.0 (which doesn't have the rotation fix for front facing camera).

molerat619, 2017-04-21:
Thanks, I have reviewed the other issues and it seems to be the same thing. Seeing that, I cannot say that it has to do something with the flash - it seemed to me like it at the beginning.

I am on 0.43 of react native and I would not like to downgrade. Is a solution for that issue on the way?

npomfret, 2017-04-24:
Image rotation is taking up a huge amount of memory (which is why the native camera doesn't do it).  My gut feeling is that it's not required any more because the metadata is no preserved in the output image.  I'll submit a PR that adds an option to `autorotate:true/false`, if someone could test it out that would be very helpful.

https://github.com/lwansbrough/react-native-camera/pull/682

closed 2018-09-14 by SimonErm