react-native-camera/react-native-camera #661: UNMET PEER DEPENDENCY react@16.0.0-alpha.6

iroshanAV, 2017-04-09:
I get this error after installing react-native-camera
c```
amerahere@0.0.1 H:\reactp\reactN\cam\camerahere
+-- UNMET PEER DEPENDENCY react@16.0.0-alpha.6
`-- react-native-camera@0.6.0 (git+https://github.com/lwansbrough/react-native-camera.git…)
npm WARN react-native-camera@0.6.0 requires a peer of react@>=15.4.0 but none was installed.
```
But then when i try to install react@16.0.0-alpha.6 i get this error
```
H:\reactp\reactN\cam\camerahere>npm install -S react@16.0.0-alpha.6
camerahere@0.0.1 H:\reactp\reactN\cam\camerahere
`-- UNMET PEER DEPENDENCY react@16.0.0-alpha.6
npm WARN react-native-camera@0.6.0 requires a peer of react@>=15.4.0 but none was installed.
npm ERR! code 1
```

iroshanAV, 2017-04-09:
do npm install react@next (or @latest) --save in project root, then try npm uninstall and npm install react-native-camera@0.6

chilion, 2017-04-10:
```
── UNMET PEER DEPENDENCY react@16.0.0-alpha.6
└── react-native-camera@0.6.0  invalid

npm WARN react-native-camera@0.6.0 requires a peer of react@>=15.4.0 but none was installed.
```

Doesn't work yet. Also your solution does not.

iroshanAV, 2017-04-10:
Reopened because some of guys having problems

@chilion hmmm May be you missed the steps. first run 
`
npm install npm@latest -g`

and then reinstall what you expected to install(because it is not yet installed)

closed 2017-04-10 by iroshanAV

litingjun2015, 2017-05-08:
DuangLink-MacBook-Pro:AwesomeProject duanglink$ npm install react@latest
AwesomeProject@0.0.1 /Users/duanglink/ReactNativeProjects/React-Native-Stable/AwesomeProject
└── (empty)

npm WARN react-native@0.44.0 requires a peer of react@16.0.0-alpha.6 but none was installed.
npm WARN react-test-renderer@16.0.0-alpha.6 requires a peer of react@^16.0.0-alpha.6 but none was installed.

iroshanAV, 2017-05-10:
@litingjun2015  did you tried to get latest version of react

litingjun2015, 2017-05-24:
@iroshanAV I'm still watching: https://github.com/lwansbrough/react-native-camera/issues/662#issuecomment-303562060

hygull, 2017-06-30:
Run the following command from your React Native project's root
```
npm install -g npm
```
then run the command which is causing the warnings/errors like below
```
...
UNMET PEER DEPENDENCY react-native-...
...
```