react-native-camera/react-native-camera #662: UNMET PEER DEPENDENCY react@16.0.0-alpha.6

chilion, 2017-04-10:
Fresh installation;

`npm install react-native-camera@0.6`
```

── UNMET PEER DEPENDENCY react@16.0.0-alpha.6
└── react-native-camera@0.6.0  invalid

npm WARN react-native-camera@0.6.0 requires a peer of react@>=15.4.0 but none was installed.
```

iroshanAV, 2017-04-10:
@chilion may be you missed the steps. first run
`
npm install npm@latest -g`

this command will make sure to get the latest version

and then reinstall what you expected to install(because it is not yet installed) and no need to uninstall anything

hudjoubert, 2017-04-13:
I had the same problema and try run this code but dont work. The problem continue.

iroshanAV, 2017-04-14:
@hudjoubert  I think it's all about npm version :/

jacobhenke, 2017-04-17:
I ran into the same issue, but was able to get it installed using yarn.

codemonkey1337, 2017-04-18:
@jacobhenke Can you please be a bit more specific on how you resolved this with yarn? I'm getting this error too, and have not been able to solve it yet.

jacobhenke, 2017-04-18:
If you have yarn installed already...

```bash
react-native init ProjectName
cd ./ProjectName
yarn add react-native-camera
npm link react-native-camera
```

When you use yarn to add react-native-camera, it DOES give you a warning about the version, but it seems to install it anyway, vs npm fails because of the version mismatch.

Hopefully this helps you!

hudjoubert, 2017-04-23:
Yea, I try do this, Really install even with notice, but when I run, the code don't work, so the problem continue.

prashantalhat, 2017-04-25:
I did
1) npm install npm@latest -g
2) npm prune
3) npm install react-native-camera
Worked for me!

prachil007, 2017-04-27:
As stated you need install latest react globally. This is not react-native-camera issue.

litingjun2015, 2017-05-08:
@prashantalhat 

`Last login: Mon May  8 14:58:50 on ttys002
DuangLink-MacBook-Pro:AwesomeProject duanglink$ react-native -v
react-native-cli: 2.0.1
react-native: 0.44.0
DuangLink-MacBook-Pro:AwesomeProject duanglink$ npm install npm@latest -g
/usr/local/bin/npm -> /usr/local/lib/node_modules/npm/bin/npm-cli.js
/usr/local/lib
└── npm@4.5.0

DuangLink-MacBook-Pro:AwesomeProject duanglink$ npm prune
npm WARN react-native-camera@0.6.0 requires a peer of react@>=15.4.0 but none was installed.
DuangLink-MacBook-Pro:AwesomeProject duanglink$ npm install react-native-camera
AwesomeProject@0.0.1 /Users/duanglink/ReactNativeProjects/React-Native-Stable/AwesomeProject
└── react-native-camera@0.6.0

npm WARN react-native-camera@0.6.0 requires a peer of react@>=15.4.0 but none was installed.`

litingjun2015, 2017-05-08:
@prachil007 

```
DuangLink-MacBook-Pro:AwesomeProject duanglink$ sudo npm install -g react@16.0.0-alpha.6
Password:
Sorry, try again.
Password:
/usr/local/lib
└── react@16.0.0-alpha.6

DuangLink-MacBook-Pro:AwesomeProject duanglink$ npm install react-native-camera
AwesomeProject@0.0.1 /Users/duanglink/ReactNativeProjects/React-Native-Stable/AwesomeProject
└── react-native-camera@0.6.0

npm WARN react-native-camera@0.6.0 requires a peer of react@>=15.4.0 but none was installed.
```

jqn, 2017-05-23:
This worked for me:
1) npm install npm@latest -g
2) npm prune
3) npm install --save react@16.0.0-alpha.6
4) npm install react-native-camera

jackson-sandland, 2017-11-04:
install with yarn worked in my case.

closed 2018-01-09 by sibelius

lauracvieira, 2018-02-14:
npm install npm@latest -g
npm prune
npm install 

worked for me! thank you for the tips!

Satish-A-Wadekar, 2020-01-30:
npm root -g
go to global packages 
open package.json
find the "peerDependencies" and add ^ on respective npm pckg 
and you are done