react-native-camera/react-native-camera #664: Make zbar optional to improve performance and quality of the scanner for android.

jqn, 2017-04-14:
### Steps to reproduce
1. Create a project for android and install react-native-camera.
2. Create another project for android and install react-native-zbar-android
3. Test them both and see the difference.

### Expected behaviour
react-native-camera for android should focus on barcode easily, handle glare or bright environments better and provide a better experience by not slowing down the device.

### Actual behaviour
react-native-camera zxing implementation has trouble focusing, reading barcodes when there is too much glare or if it's too bright outside. In my test with react-native-zbar-android I didn't experience any of these issues and didn't slow down the device.

### Environment
- **Node.js version**:
- **React Native version**: 0.40
- **React Native platform + platform version**:  Android 6.0

### react-native-camera
**Version**:  "master"

react-native-zbar-android is not complete yet but review this repo for an improved android implementation of the zbar for android:

https://github.com/dm77/barcodescanner

tonybebber, 2017-04-14:
+1  - The current scanning capabilities work great on iOS! However on Android, Zbar does seem to scan faster. I'm assuming this wouldn't be trivial, but how difficult would it be to integrate the option to initialize the scanner w/ either scanning library (zxing or zbar)?

mgoicoechea, 2017-04-19:
I have worked with Android scanners for years for automotive applications.  zxing is weak.  The native iOs scanner blows it out of the water.  The only library that comes close is zbar.

dinhha, 2017-06-21:
+1000

kevinnguy, 2017-09-12:
Do you all know if there are any other react native alternatives for barcode scanning?

samhv, 2017-12-14:
Zxing vs Zbar vs Google Barcode

Zxing is a bad idea today because there is no android specific library that manages the camera APIs, then we need to copy-paste and hack it for work (difficult to maintain in a current v0.X). Zbar is ok but is poor in quality (often misread the numbers). Google Barcode is the best in almost all but it need the Mobile Vision library downloaded in the device.

in my projects react-native-camera work fine in ios, for android I have https://github.com/ekreutz/react-native-barcode-scanner-google and it works excellently.

fenglu09, 2017-12-26:
@samhv   In china, `react-native-barcode-scanner-google ` not working. will got `no_play_service` error

jqn, 2017-12-26:
@fenglu09 are you testing on a simulator or a physical device?

fenglu09, 2017-12-27:
@jqn  On a physical device. `Meizu M5 Note`， Android 7.0.

sibelius, 2018-03-16:
@DavidRajPaul can u review this https://github.com/react-native-community/react-native-camera/pull/862

let us if this is good, and we can merge it