react-native-camera/react-native-camera #665: Able to QR-scan image from gallery?

kylanhurt, 2017-04-14:
I'm using version 0.24.1 and I am curious if it's possible to pass an image from the gallery (I'm using react-native-image-picker) to react-native-camera to be scanned as a QR-code, rather than live-scanning via the camera.  
Any help would be appreciated.

yellowJhon, 2017-09-05:
Hi, bother, have you solved this problem?

Obooman, 2017-09-27:
though "scan QRcode from local image" is not what `react-native-camera` should support, but we really can not find a library can recognize QRcode as sensitive as this library do.

split QRcode decode part to a standalone package or serve a solution would be appreciated.

sibelius, 2018-01-09:
this should be another package

closed 2018-01-09 by sibelius

0xcodezero, 2018-03-07:
@kylanhurt Did you Find a way to implement Image scanning from Gallery that contains QR code?

yellowJhon, 2018-03-07:
You May extends this library by add a @ReactMothod to recognize local image.

0xcodezero, 2018-03-07:
@yellowJhon Do you suggest another library to do so, Or I could pass a frame instead of using the video live feed of react-navtive-camera, or writing a method to scan the image manually?

I have also explored the repo branches & issues, there is guidance to include such functionality, so your guidance is highly appreciated 👍

yellowJhon, 2018-03-07:
Yeah. 
For example, react-navtive-camera use the package of com.google.zxing to  scan the image.
Diving into the source code:
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/tasks/BarCodeScannerAsyncTask.java

You can see the recognize task of this library also pass a bitmap (a frame of video) as the input, and out the result.

SO, you MAY learn from his implemetion to achieve your function