react-native-camera/react-native-camera #666: Scan Code Not Working Properly

nhuthuynh, 2017-04-16:
### Steps to reproduce
Scanner function not working properly with barcode image, same image but different url link have difference result.

```
class ScanView extends React.Component {
    props: {
        onScanned: ()=>void;
        navigator: Navigator;
        isLoading: boolean;
        isScannable: boolean;
    }

    constructor(props) {
        super(props);
    }

    render(){
        const {scene} = this.props;
        const {isLoading} = this.props;
        const ableToScan = isLoading ? null : this.props.onScanned;
        const camera = !isLoading && scene ? <Camera
            ref={(cam) => {
                this.camera = cam;
            }}
            style={ styles.camera }
            aspect={Camera.constants.Aspect.fill}
            onBarCodeRead={ableToScan}
            barCodeTypes={barCodeTypes}
            captureTarget={Camera.constants.CaptureTarget.memory}
        >
        </Camera> : null;
        return(
            <View style={styles.container}>
                <View style={styles.cameraContainer}>
                    <View style={styles.cameraBorderHorizontal}>
                        {camera}
                    </View>
                    <View style={[styles.cameraBorder, styles.cameraBorderLeft]}></View>
                    <View style={[styles.cameraBorder, styles.cameraBorderRight]}></View>
                </View>
            </View>
        )
    }
}

const barCodeTypes = Platform.select({
    ios:['org.iso.Aztec',
        'org.iso.Code128',
        // 'code138' should be here, but not sure of the namespace (found in code)
        'org.iso.Code39',
        'org.iso.Code39Mod43',
        'com.intermec.Code93',
        // 'datamatrix' should be here, but not sure of the namespace (found in code)
        'org.gs1.EAN-13',
        'org.gs1.EAN-8',
        // 'interleaved2of5' should be here, but not sure of the namespace (found in code)
        // 'itf14' should be here, but not sure of the namespace (found in code)
        'org.iso.PDF417',
        'org.iso.QRCode',
        'org.gs1.UPC-E'],
    android: ['aztec',
      'codabar',  // Not in the docs, but found in code
      'code128',
      'code39',
      // 'code39mod43',  // Found in the docs but not in the code
      'code93',
      'datamatrix',  // When available(?), according to the docs
      'ean13',
      'ean8',
      'interleaved2of5',  // When available(?), according to the docs
      // 'itf14',  // When available(?), according to the docs, but not found in code
      'maxicode',  // Not in the docs, but found in code
      'pdf417',
      'qr',
      'rss14',  // Not in the docs, but found in code
      'rssexpanded',  // Not in the docs, but found in code
      'upca',
      'upce',
      'upceanextension']
})

const styles = StyleSheet.create({
    container: {
        flex: 1,
        backgroundColor: '#f5f5f5',
        alignItems: 'center',
        justifyContent: 'center',
    },
    cameraContainer: {
        ...StyleDefault.cameraContainer
    },
    camera: {
        flex: 1
    },
    cameraBorderHorizontal: {
        ...StyleDefault.cameraBorderHorizontal
    },
    cameraBorder: {
        ...StyleDefault.cameraBorder
    },
    cameraBorderLeft: {
        left: 0
    },
    cameraBorderRight: {
        right: 0
    }
})

export default ScanView
```

### Expected behaviour
Scan function should working with both image.

### Actual behaviour

Scan working with this link of image:
https://www.google.com.vn/search?q=barcode&espv=2&source=lnms&tbm=isch&sa=X&ved=0ahUKEwiNtomTu6jTAhXEWLwKHUL8AJoQ_AUIBigB&biw=1385&bih=759#imgrc=eERSULOTr903WM:

But not working with this link:
http://www.gs1.org/sites/default/files/docs/barcodes/EAN-13.png

### Environment
- **Node.js version**:
4.2.6
- **React Native version**:
react-native-cli: 2.0.1
react-native: 0.42.0
- **React Native platform + platform version**:
nexus 5, android 6
### react-native-camera
**Version**: npm version or "master"
"react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git"

kylanhurt, 2017-04-17:
So out of curiosity how are you scanning the images? Sending them into the JS as actual URL's or physically scanning images on your computer with a phone, etc?

nhuthuynh, 2017-04-17:
@kylanhurt i built signed apk file, installed on the nexus 5 android 6 and test with those links.

closed 2018-09-14 by SimonErm