react-native-camera/react-native-camera #667: mirrorImage prop is ignored on iOS

davidkpham, 2017-04-16:
### Steps to reproduce
1. Set mirrorImage prop to true on Camera component

### Expected behaviour
Resulting video should be mirrored

### Actual behaviour
Resulting video is not mirrored

### Environment
- **Node.js version**: 7.7.1
- **React Native version**: 0.43.3
- **React Native platform + platform version**: iOS 10.3

### react-native-camera
**Version**: master

I've tried debugging by going straight to the RCTCameraManager class and putting in some logging statements but I'm not seeing anything printed out. Has anyone noticed this? In the meantime - I'll keep digging.

davidkpham, 2017-04-16:
After some more digging, it looks like the mirror transformation is only applied to stills, not videos.

daltonconley, 2017-06-10:
+1 would love to have recorded videos mirrored.

Liron-K, 2017-11-02:
I had an issue with the `mirrorImage` prop on iOS unless I also set `fixOrientation` or `rotation`.

This is because the code for setting the mirror happens in `newCGImageRotatedByAngle` which only gets called if one of the above props is set.

RCTCameraManager line 606.

I tried to add an if statement there to also call `newCGImageRotatedByAngle` if mirrorImage is set, but that mirrored vertically instead of horizontally, and I ended up getting the result I wanted, by setting both `mirrorImage` and `fixOrientation`.

arapocket, 2017-12-01:
@Liron-K It's also being ignored for me with stills but i'll try out what you said.

sibelius, 2017-12-10:
we gonna accept a PR to handle this

geniuscd, 2018-02-17:
I came across this issue, taking a photo with mirrorImage prop to true is being ignored and when taking a selfie, the image is flipped. so I read the Objective-c (iOS) code, and found out that we have to pass the mirrorImage in the dictionary object passed to takePicture method. [line code](https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L324)

so instead of adding it as a prop in the RNCamera component, we should supply it as an option to the `takePictureAsync(options)` method

a sample code below to demonstrate 

`const options = { quality: 0.5, base64: true, mirrorImage: true };`
`const data = await this.camera.takePictureAsync(options)`

this is only tested on iOS, not android.

hangindan, 2019-12-18:
Anyone else still having issues on iOS? The mirrorImage option does not appear to have an affect. Tried setting to true and false (along with fixOriention). Back camera appears to work fine regardless of setting selfie mode always flips on the vertical. 

RNCamera 3.15.0
React-native: 0.61.3

closed 2020-02-24 by stale[bot]

tayguohong, 2020-02-27:
Why is this issue closed? It is not fixed yet. Both the front and rear camera get mirrored and rotated although fixOrientation and mirrorImage props being set.