react-native-camera/react-native-camera #671: Camera　interface white SDK: 22, the user manually shut down Camera permissions

TrustTheBoy, 2017-04-19:
Camera authority is not open, the camera interface is white, how to solve?
PermissionsAndroid only works at SDK> = 23

windows 10
phone: Samsung
Android ：5.0

ptomasroos, 2017-04-26:
We're handling this out side the component in our app. I would suggest you to do the same, check for permissions and do not show the view if someone didnt grant permission

pvanliefland, 2017-06-07:
@TrustTheBoy see my comment here: #224

lhang, 2017-08-26:
@ptomasroos I tried to use the PermissionAndroid to check camera permission, but in some phone like meizu, oneplus, check function return permission granted result is true after I close the permission. some phone will crash if I close the camera permission.

closed 2018-09-14 by SimonErm