react-native-camera/react-native-camera #672: Third party Android camera interface

skovhus, 2017-04-20:
A thought. 💭 

Seeing the many tickets related to rotation and memory issues on Android, did you consider leveraging libraries like https://github.com/google/cameraview or https://github.com/gogopop/CameraKit-Android as a base implementation for Android?

PRO: Ease the work of maintaining this library and would make this work here more focused around nice React Native packaging.

CONS: Another layer of abstraction (but seems like a fairly solid one, also supporting Camera2 API).

ptomasroos, 2017-04-26:
Find me some alternatives which has the same feature set for IOS and I will write you a package in  the weekend :)

npomfret, 2017-04-26:
I think he means use one of these as the basis for the android implementation in place of one of the editing android code

ptomasroos, 2017-04-26:
Well with all cross cutting concerns the api surfaces starts to become quite hard to maintain, yes def harder on Android.

Bar code scanning for instance, who of the maintainers use that in their production apps these days? We dont, hard to support, there's a lot of issues concerning those as well. 

Video, people expect mirroring, encoding, ect think we def need to nail out a larger roadmap in order to settle and expose a better package with less problems for the end users.

And yes, rotation mirroring ect should be a solved problem already I agree.

skovhus, 2017-04-26:
This only means using an Android lib as basis for the implementation. The iOS version of this library seems more stable. 

But yes. Wanting to do video and bar code scanning might complicate things...

npomfret, 2017-04-27:
A friend of mine wrote a camera app called NovaCamera for both iOS and android, it's excellent I have to admit - worth checking out.  Not as fully featured maybe (no bar code scanning, no video) but lots of stuff around exposure, focus and iso control.  But maybe we could learn from some of the stuff he did:

https://github.com/novaphotos/nova-ios-app/blob/develop/NovaCamera/Services/SSCaptureSessionManager.m#L280

https://github.com/novaphotos/nova-android-opencamera/blob/master/src/com/almalence/opencam/cameracontroller/CameraController.java#L3282

closed 2018-09-14 by SimonErm