react-native-camera/react-native-camera #673: Error creating media file.

Borisboky, 2017-04-21:
I'm getting an  **Error creating media file** error on android simulator when I try to take a picture.

My code is as follows:

```
takePicture() {
        this.camera.capture({target: Camera.constants.CaptureTarget.disk})
            .then(data => this.props.pictureTaken(data);)
            .catch(err => console.error(err));
    }

render() {
        return (
            <View style={styles.container}>
                <Camera
                    ref={(cam) => {
                            this.camera = cam;
                          }}
                    style={styles.preview}
                    aspect={Camera.constants.Aspect.Fill}>
                </Camera>

                <View style={styles.footer}>
                    <View style={styles.footerButton}>
                        <TouchableHighlight onPress={this.takePicture.bind(this)} underlayColor='#ff1c4d'>
                            <View style={styles.contentFooterButton}>
                                <Icon
                                    name='ios-camera-outline'
                                    size={25}
                                    style={styles.icon}/>
                                <Text style={styles.footerButtonText}>Picture</Text>
                            </View>
                        </TouchableHighlight>
                    </View>
                </View>
            </View>
        );
    }       
```

When takePicture function is called I get the error.

I use : 

- node v6.9.1
- react-native: 0.41.2
- Android simulator : Android 6.0

package.json : 

- _"react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git",_

diegocouto, 2017-04-21:
Hi, @Borisboky!

Are you sure you have updated your AndroidManifest file with the right permissions?

```
<uses-permission android:name="android.permission.RECORD_AUDIO"/>
<uses-permission android:name="android.permission.RECORD_VIDEO"/>
<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
```

Borisboky, 2017-04-21:
Hi @diegocouto,

Yes, I'm sure. See photo : 
[image]

npomfret, 2017-04-24:
The simulator doesn't have a camera... does it!?

Borisboky, 2017-04-24:
@npomfret It's the same problem on the device. The app crashes if the takePicture() function gets called.

npomfret, 2017-04-24:
ok, do you have a stack trace from logcat? I suspect its running out of memory trying to rotate the image.  If so, would you mind testing my PR which possibly works around the issue?

https://github.com/lwansbrough/react-native-camera/pull/682

Borisboky, 2017-04-25:
With `react-native log-android` I get 
[image]

Thus, 

`04-25 09:23:47.608  2394  2550 E ReactNativeJS: { [Error: Error creating media file.] framesToPop: 1, code: 'EUNSPECIFIED' }` is what I get when the function takePicture() gets called.

npomfret, 2017-04-25:
Hmm, there does seem to be enough information there.  The output of logcat is usually much more verbose, and when your app crashes there's usually  a stack trace.  I can't think of a way to help you without that information.  Have you tried running it directly from android studio?  The Android Montitor tab shows the full log cat from your device.

Borisboky, 2017-04-25:
If I show all error using `adb logcat "*:E"` I get 

```
Boriss-MacBook-Pro:autralis-mobile boris$ adb logcat "*:E"
--------- beginning of main
--------- beginning of system
04-25 09:44:30.911  1229  1229 E         : batterystats service unavailable!
04-25 09:44:30.912  1229  1229 E         : batterystats service unavailable!
04-25 09:44:30.914  1229  1229 E         : batterystats service unavailable!
04-25 09:44:30.914  1229  1229 E         : batterystats service unavailable!
04-25 09:44:30.975  1186  1186 E libEGL  : load_driver(/system/lib64/egl/libGLES_emulation.so): dlopen failed: library "/system/lib64/egl/libGLES_emulation.so" not found
04-25 09:44:31.098  1186  1186 E SurfaceFlinger: hwcomposer module not found
04-25 09:44:31.110  1186  1186 E cutils-trace: Error opening trace file: Permission denied (13)
04-25 09:44:31.327  1186  1243 E SurfaceFlinger: ro.sf.lcd_density must be defined as a build property
04-25 09:44:31.380  1261  1268 E libEGL  : load_driver(/system/lib64/egl/libGLES_emulation.so): dlopen failed: library "/system/lib64/egl/libGLES_emulation.so" not found
04-25 09:44:31.409  1261  1268 E cutils-trace: Error opening trace file: Permission denied (13)
04-25 09:44:31.932  1229  1229 E cutils-trace: Error opening trace file: Permission denied (13)
04-25 09:44:31.936  1229  1229 E CameraService: setUpVendorTags: Vendor tag operations not fully defined. Ignoring definitions.
04-25 09:44:33.148  1232  1232 E libEGL  : load_driver(/system/lib64/egl/libGLES_emulation.so): dlopen failed: library "/system/lib64/egl/libGLES_emulation.so" not found
04-25 09:44:33.162  1233  1233 E libEGL  : load_driver(/system/lib/egl/libGLES_emulation.so): dlopen failed: library "/system/lib/egl/libGLES_emulation.so" not found
04-25 09:44:33.170  1232  1232 E Hyphenator: error loading hyphenation /system/usr/hyphen-data/hyph-en-us.pat.txt
04-25 09:44:33.170  1232  1232 E Hyphenator: java.io.FileNotFoundException: /system/usr/hyphen-data/hyph-en-us.pat.txt: open failed: ENOENT (No such file or directory)
04-25 09:44:33.170  1232  1232 E Hyphenator: 	at libcore.io.IoBridge.open(IoBridge.java:452)
04-25 09:44:33.170  1232  1232 E Hyphenator: 	at libcore.io.IoUtils$FileReader.<init>(IoUtils.java:207)
04-25 09:44:33.170  1232  1232 E Hyphenator: 	at libcore.io.IoUtils.readFileAsString(IoUtils.java:114)
04-25 09:44:33.170  1232  1232 E Hyphenator: 	at android.text.Hyphenator.loadHyphenator(Hyphenator.java:96)
04-25 09:44:33.170  1232  1232 E Hyphenator: 	at android.text.Hyphenator.init(Hyphenator.java:154)
04-25 09:44:33.170  1232  1232 E Hyphenator: 	at com.android.internal.os.ZygoteInit.preloadTextResources(ZygoteInit.java:207)
04-25 09:44:33.170  1232  1232 E Hyphenator: 	at com.android.internal.os.ZygoteInit.preload(ZygoteInit.java:186)
04-25 09:44:33.170  1232  1232 E Hyphenator: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:593)
04-25 09:44:33.170  1232  1232 E Hyphenator: Caused by: android.system.ErrnoException: open failed: ENOENT (No such file or directory)
04-25 09:44:33.170  1232  1232 E Hyphenator: 	at libcore.io.Posix.open(Native Method)
04-25 09:44:33.170  1232  1232 E Hyphenator: 	at libcore.io.BlockGuardOs.open(BlockGuardOs.java:186)
04-25 09:44:33.170  1232  1232 E Hyphenator: 	at libcore.io.IoBridge.open(IoBridge.java:438)
04-25 09:44:33.170  1232  1232 E Hyphenator: 	... 7 more
04-25 09:44:33.171  1232  1232 E Hyphenator: error loading hyphenation /system/usr/hyphen-data/hyph-eu.pat.txt
04-25 09:44:33.171  1232  1232 E Hyphenator: java.io.FileNotFoundException: /system/usr/hyphen-data/hyph-eu.pat.txt: open failed: ENOENT (No such file or directory)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at libcore.io.IoBridge.open(IoBridge.java:452)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at libcore.io.IoUtils$FileReader.<init>(IoUtils.java:207)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at libcore.io.IoUtils.readFileAsString(IoUtils.java:114)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at android.text.Hyphenator.loadHyphenator(Hyphenator.java:96)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at android.text.Hyphenator.init(Hyphenator.java:154)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at com.android.internal.os.ZygoteInit.preloadTextResources(ZygoteInit.java:207)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at com.android.internal.os.ZygoteInit.preload(ZygoteInit.java:186)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:593)
04-25 09:44:33.171  1232  1232 E Hyphenator: Caused by: android.system.ErrnoException: open failed: ENOENT (No such file or directory)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at libcore.io.Posix.open(Native Method)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at libcore.io.BlockGuardOs.open(BlockGuardOs.java:186)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at libcore.io.IoBridge.open(IoBridge.java:438)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	... 7 more
04-25 09:44:33.171  1232  1232 E Hyphenator: error loading hyphenation /system/usr/hyphen-data/hyph-hu.pat.txt
04-25 09:44:33.171  1232  1232 E Hyphenator: java.io.FileNotFoundException: /system/usr/hyphen-data/hyph-hu.pat.txt: open failed: ENOENT (No such file or directory)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at libcore.io.IoBridge.open(IoBridge.java:452)
04-25 09:44:33.171  1232  1232 E Hyphenator: 	at libcore.io.IoUtils$FileReader.<init>(IoUtils.java:207)
[... cut ...]
04-25 09:48:23.907  1229  1765 E CameraService: disconnect: Disconnect called on already disconnected client for device 0
04-25 09:48:30.093  1828  2593 E Surface : getSlotFromBufferLocked: unknown buffer: 0x7f4bd0d5aa20
04-25 09:48:33.242  1229  1579 E CameraService: disconnect: Disconnect called on already disconnected client for device 0
04-25 09:48:37.291  2536  2550 E RCTCameraModule: failed to create directory:/sdcard/Pictures
04-25 09:48:37.293  2536  2686 E ReactNativeJS: { [Error: Error creating media file.] framesToPop: 1, code: 'EUNSPECIFIED' }
```

npomfret, 2017-04-25:
There are an awful lot of FileNotFound exceptions in your log.  That doesn't look healthy.  Is the device working properly?  Have you tried creating a file on it using something like react-native-fs?

Borisboky, 2017-04-25:
Yes. Everything else works fine. I used react-native-fs but I deleted it and I don't use it anymore.

Borisboky, 2017-04-26:
@npomfret I think that I found what the problem was. The `buildToolsVersion` in the `build.gradle` file was **"23.0.2"** I just changed it manually to **"25.0.3"** and now it works.

closed 2017-04-27 by npomfret

nicgutierrez, 2017-05-18:
Hey, I'm having the exact same issue but changing the buildToolsVersion to 25.0.3 didn't make any difference. Still failing. Any ideas?

npomfret, 2017-05-21:
Again, can you paste the stack trance in here.  I'm surprised changing the tools version has fixed it, that doesn't make much sense to me at the moment.