react-native-camera/react-native-camera #675: Tap to focus don't work on Android (And auto focus is working badly)

Thomeuxe, 2017-04-22:
### Steps to reproduce
1. Here's the component of my view (there's only that component) : 

```
<Camera
         ref={(cam) => {
         this.camera = cam;
         }}
         captureQuality={Camera.constants.CaptureQuality["720p"]}
         captureTarget={Camera.constants.CaptureTarget.memory}
         defaultOnFocusComponent={true}
         captureAudio = {false}
         onFocusChanged={() => {console.log("focusChanged");}}
         onZoomChanged={() => {console.log("zoomChanged");}}
         style={styles.preview}
         aspect={Camera.constants.Aspect.fill}/>
```

### Expected behaviour
On tap --> focus
On move --> try autofocus

Any idea ?

### Actual behaviour
On tap --> Nothing
On move --> autofocus sometime happens, sometime don't

### Environment
- **Node.js version**: 7.6.0
- **React Native version**: 0.42.0
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: 0.6.0 or 0.7.0, same issue

mikaello, 2017-04-24:
This issue is maybe solved in PR  https://github.com/lwansbrough/react-native-camera/pull/575 ?

jarredszabadi, 2017-04-27:
Upgraded to latest version today and same thing. 
Currently running:
| software         | version
| ---------------- | -------
| react-native-camera |  0.7.0
| react-native     |  0.43.4
| node             | v7.8.0
| npm or yarn      | npm 4.5.0

Previously I was running react-native 0.42.3 with the forked version of this repo with the fix and it was working great.

mikaello, 2017-04-28:
react-native-camera 0.7.0 was released before https://github.com/lwansbrough/react-native-camera/pull/578 was merged, so I guess you have to wait for the next release, or use the master branch as a dependency in your package.json: `"react-native-camera": "lwansbrough/react-native-camera"`

jarredszabadi, 2017-04-28:
Ah! Of course! Thank you.

escobar5, 2017-05-04:
I've just tested it in 0.8.0 and having the same problems, should that be fixed already in that version?

csae0, 2017-10-04:
add this to AndroidManifest:

<uses-permission android:name="android.permission.CAMERA" />
<uses-feature android:name="android.hardware.camera" />
<uses-feature android:name="android.hardware.camera.autofocus" />

For me, it seems autofocus works better now

brooth, 2017-11-21:
In my case it happened when I updated the component (`setState()`) right after `camera.capture()`. I removed `<Camera/>` in order to show `<ActivityIndicator/>`. So it seems that it get stuck if it's saving the image but the component doesn't exist anymore

iarambula, 2018-05-16:
Shouldn't @csae0's comment be in the documentation?

n1ru4l, 2018-12-19:
@iarambula Feel free to send a pull request that improves the documentation.

I am going to close this issue. Please open a new or search for an existing open issue if you are still experiencing this issue with `RNCamera`.

closed 2018-12-19 by n1ru4l