react-native-camera/react-native-camera #680: React 16 support?

kesha-antonov, 2017-04-24:
Hello!

RN 0.43 require to use react-16.

It'll be great to update this lib to use React 16.
Can we do that?

https://github.com/lwansbrough/react-native-camera/issues/662

tamalsen, 2017-04-25:
Started upgrading. But having some issue with react-native-maps with @16.*.*

sebandres, 2017-07-28:
@kesha-antonov @tamalsen Just did this [PR](https://github.com/lwansbrough/react-native-camera/pull/811) to fix the PropTypes compatibility with React 15.5+

Once it is merged you should be able to use React 15.5+ (using 16 beta2 now)

sebandres, 2018-01-09:
@sibelius @rt2zz  shouldn't this be closed as per the comments on this [PR](https://github.com/react-native-community/react-native-camera/pull/811)?

tamalsen, 2018-01-09:
I think you should go ahead and close this. @sibelius

closed 2018-01-09 by sibelius