react-native-camera/react-native-camera #681: onBarCodeRead doesn't get called

MathiasDeWeerdt, 2017-04-24:
### Steps to reproduce
1. Trying to detect and scan barcodes
2. Copy code from the repo
3. Test the app

### Code
`import React, { Component } from 'react';
import {
  AppRegistry,
  StyleSheet,
  Text,
  View,
  Alert
} from 'react-native';
import Camera from 'react-native-camera';

export default class funapp extends Component {

  takePicture() {
    const options = {};
    this.camera.capture({metadata: options})
      .then((data) => console.log(data))
      .catch(err => console.error(err));
  }

  onBarCodeRead(data) {
    Alert.alert("Test", "Test");
    console.log("Barcode: " + data);
  }

  render() {
    return (
      <View style={styles.container}>
        <Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={styles.preview}
          aspect={Camera.constants.Aspect.fill}
          onBarCodeRead={this.onBarCodeRead.bind(this)}
          barCodeTypes={[
          	Camera.constants.BarCodeType.upce,
          	Camera.constants.BarCodeType.ean8,
          	Camera.constants.BarCodeType.ean13,
          ]}>
          <Text style={styles.capture} onPress={this.takePicture.bind(this)}>[Take picture]</Text>
        </Camera>
      </View>
    );
  }
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    flexDirection: 'row',
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center'
  },
  capture: {
    flex: 0,
    backgroundColor: '#fff',
    borderRadius: 5,
    color: '#000',
    padding: 10,
    margin: 40
  }
});

AppRegistry.registerComponent('funapp', () => funapp);
`

### Expected behaviour
Call the function: onBarCodeRead

### Actual behaviour
Nothing, not even an error. 

### Environment
- **Node.js version**: V7.9.0
- **React Native version**: 0.43.4
- **React Native platform + platform version**: Android 7

### react-native-camera
**Version**: 4.5.0

Fconstant, 2017-04-24:
Same with me on iOS 10.3.1

yuguangmao, 2017-05-08:
[](https://github.com/moaazsidat/react-native-qrcode-scanner)
this module use 'react-native-camera' ,it works!

visoft, 2017-06-21:
I see the same thing on Android 6.0 where nothing happens when scanning a barcode. It works completely fine in iOS. I tried stripping out the `barCodeTypes` to see if that was the problem, but it wasn't. `onBarCodeRead` never seems to be called.

Node.js version v8.1.0
react-native: v0.45.1
react-native-camera: GitHub master

visoft, 2017-06-27:
@yuguangmao I've had some success scanning QR codes with react-native-camera, but not EAN13s

vagnerSl, 2017-07-24:
+1 IOS 10.3.2

igroz777, 2017-07-31:
Same here. When using an emulator it works fine but when i deploy to a real device function doesn't get called

maikenmadsen92, 2017-08-01:
have someone figured out the answear for this question? :-)

WANGjieJacques, 2017-09-13:
for the android version, I finally found the bug, 
```
@Override
        public void onSensorChanged(SensorEvent event) {
            float x = event.values[0];
            float y = event.values[1];

            if (x<5 && x>-5 && y > 5)
                mOrientation = Surface.ROTATION_0; // portrait
            else if (x<-5 && y<5 && y>-5)
                mOrientation = Surface.ROTATION_270; // right
            else if (x<5 && x>-5 && y<-5)
                mOrientation = Surface.ROTATION_180; // upside down
            else if (x>5 && y<5 && y>-5)
                mOrientation = Surface.ROTATION_90; // left

            if (mListener != null) {
                mListener.orientationEvent();
            }
        }
```
every time I start the camera, if the device has been moved, it will get a TYPE_ACCELEROMETER sensorEvent, and the `onSensorChanged` will be called, and finally the `captureWithOrientation` will be called, in this method, `camera.setPreviewCallback(null);`, the preview call back will be set to null, so the `reactContext.getJSModule(DeviceEventManagerModule.RCTDeviceEventEmitter.class).emit("CameraBarCodeReadAndroid", event);` won't be called.
I don't know what's the perfect solution for this, I just remove `camera.setPreviewCallback(null);` in the method `private void captureWithOrientation(final ReadableMap options, final Promise promise, int deviceOrientation)`

csumrell, 2017-10-16:
I am still having this problem on older android versions (5.1.1) but it works on newer ones and IOS. I tried the bug fix @WANGjieJacques mentioned above but it did not fix the issue for me

WANGjieJacques, 2017-10-17:
hi @csumrell try to set a breakpoint in the line `camera.setPreviewCallback(null)` to see if the callback has been set to null.

dann1609, 2018-01-23:
for me on android work without problem, but in iOS doesn't work.

SimonErm, 2018-10-11:
RCTCamer is deprecated, use RNCamera instead. If this error occures with RNCamera too, please open a new issue.

closed 2018-10-11 by SimonErm