react-native-camera/react-native-camera #683: There is no "bounds" property on returned info.

MocanuMarius, 2017-04-24:
Hi, the bounds property is missing on android. Can we please get some answer to why that is so ? At least put a disclaimer there that it's not working.

Except that property, everthing else is in there.

### Environment
- **Node.js version**: 3.10.10
- **React Native version**: 0.42.3 (CLI 2.0.1)
- **React Native platform + platform version**: Android 5.1

### react-native-camera
**Version**: 0.7.0

MocanuMarius, 2017-04-24:
After reviewing your android files i see that there is no real implementation of this using the Android Barcode api which should support this .. it's using some other api written by someone. Am i wrong here ?

MocanuMarius, 2017-04-24:
Ok i have found the problem and the actual Zhing api or whatever .. i've put this piece of code in there
[image]

Now i'm gonna see if i manage to rebuild this library of yours somehow... npm scripts are : [] :/

MocanuMarius, 2017-04-24:
After spending the most usless hour in my programming life fiddling with a simple serialization of an Array in a stupid language, i have come to this solution that works: I will post my fork in a link below somewhere 
[image]

MocanuMarius, 2017-04-24:
https://github.com/MocanuMarius/react-native-camera

MocanuMarius, 2017-04-24:
After playing with it a little bit i realised that yes it works .. but the order those points are received at the x-y pairing they have is completely hard to understand ... :/ they are not corner points basically ,  they're just basically mingled togheter. Also keep in mind you have to use PixelRatio to get real screen space by dividing it from those points bounds.

MocanuMarius, 2017-04-25:
.....
```
	readQR( e ) {
		let bounds = JSON.parse(e.bounds).map((bigArr) => bigArr.map(
			(item, index) => {
				// Some random width ratio that works for me
				if ( index == 0 ) return Math.floor(item/3)
				else return Math.floor(item/4)
				// Some random height ration that works for me
			}
		))
```

					<Image
						style={{
						opacity: this.state.opacity,
						height: this.state.bounds[0][1] - this.state.bounds[1][1],
						width: this.state.bounds[2][0] - this.state.bounds[1][0],
						top: this.state.bounds[1][1]*0.7,
						left: this.state.bounds[1][0] - 16,
						resizeMode: 'contain',
						position: 'absolute'
					}}
						source={{
						uri: this.state.parsedUrl
					}}/>

MocanuMarius, 2017-04-25:
[image]

ptomasroos, 2017-04-26:
Will close since you consider it closed. And #631 already exists

MocanuMarius, 2017-04-26:
I wouldn't close this... i wanted to leave it open to have a discussion on how to implement this in a PR.. i could make the PR for you but i need a review

ptomasroos, 2017-04-27:
Sure thing.

MocanuMarius, 2017-04-27:
@ptomasroos is it ok if we have a different bounds for android and iOS ? I would rather export the raw points given to me than calculating centers and origin and things like that you have for iOS. Does any of the maintainers have experience with Java ?

gabox8888, 2017-07-09:
I feel like you are adding unnecessary serialization code. This should suffice 

event.putString("bounds", java.util.Arrays.toString(result.getResultPoints()));

assafb81, 2017-07-17:
I did not understand how to add/set 'bound' property.
I want to do something like in the attached file - to focus the user on  a small box instead of entire screen
the result should look something like:

[image]

MocanuMarius, 2017-07-17:
@assafb81 My print isn't focusing one region of the camera, it's detecting qr-codes everywhere. How you could do what you want though is by detecting codes everywhere, but only reacting when codes are within a certain bound in your camera view. How i would do this: 
1. Figure out the position and bounds of your "ok" box in screen-space.
2. Figure out the position on screen for your QR Code regardless of where it is ( like i did in this issue )
3. Only react if the qr is dected within the 1) bounds, using some maths.

darioodiaz, 2017-07-24:
I was looking that issue, no bounds property on event, that solution is on a PR already? Amazing job boys!

MocanuMarius, 2017-07-24:
@darioodiaz hey thanks..but since i'm not using this module anymore, (test project) ... someone else can pick up my code / PR ...someone with some Java knowledge prefferably and make a decent PR ..maybe try to sync with the other IOS interface. Until then...i guess what i provided above is a hack solution that works.

darioodiaz, 2017-07-24:
@MocanuMarius until wednesday I'm busy but after that I'll try to do a PR @ptomasroos that feature are solved? or will be available in the next release? If not I'll check and try to do a PR

blankg, 2017-08-15:
Would like to propose the following code:
```
                WritableArray resultPoints = Arguments.createArray();
                ResultPoint[] points = result.getResultPoints();
                if(points != null && points.length > 0) {
                    int pointsLength = points.length;
                    for(int index = 0; index < pointsLength; index++) {
                        ResultPoint point = points[index];
                        if(point != null) {
                            WritableMap newPoint = Arguments.createMap();
                            newPoint.putString("x", String.valueOf(point.getX()));
                            newPoint.putString("y", String.valueOf(point.getY()));
                            resultPoints.pushMap(newPoint);
                        }
                    }
                }
               event.putArray("bounds", resultPoints);

```
The result you will get in the javascript side would be an array of objects e.g. [{x: "222", y: "333"}, ...]

blankg, 2017-08-17:
@darioodiaz, are you still planning to do a PR? if not let me know and I can do it.

darioodiaz, 2017-08-17:
Hi everybody! I was and I'm too busy with my job and studies so I haven't time to do this, @blankg if you want to do the PR do it! I have much personal stuffs to do :( sorry boys

ptomasroos, 2017-08-25:
Fixed in #845

closed 2017-08-25 by ptomasroos