react-native-camera/react-native-camera #686: How to set the Video recoding timing ????

farrukhbashir, 2017-04-28:
### Steps to reproduce
1.Open Camera 
2.start recording 
3.After this i want to set the limited time for video recording after this time video automatically save

### Expected behaviour
Video Save

### Actual behaviour
I have no code for this implementation plzz anyone help me how i implement this 

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

Symous, 2017-05-09:
Maybe you can use setTimeout to stop and save automatically if the user doesn't stop it manually.

```js
setTimeout(()=>{
     //Stop and save video here.
},10000)
```

Symous, 2017-05-11:
U also can add totalSeconds option when we invoke capture() in video mode , it will invoke stopCapture() after the totalSeconds. :-D

farrukhbashir, 2017-05-11:
ok......after the video complete when i try to capture pic then app stop working???

y00rb, 2018-01-02:
@Symous why can't get video`data` when call `camera.stopCapture()` ? I following the sample codes. but I using codes like:
```
      this.camera.capture({mode: Camera.constants.CaptureMode.video})
          .then((data) => console.log(data))
```
I can't see anything output in console log

arapocket, 2018-03-02:
The totalSeconds option doesn't seem to do anything for me.

closed 2018-03-03 by sibelius