react-native-camera/react-native-camera #687: Feature request: Add frame, audio and video encoding bit rate options

diegocouto, 2017-04-30:
Hi there,

A 10s long video with `480p` quality has a file size of more or less 3MB. For a lot of use cases, this is too large and this is why I think it would be great to allow some control over frame, audio and video encoding bit rates.

**Android Support**

I have updated **RCTCameraModule.java** with these hardcoded configs as an example, and we are getting files at least 3X smaller:
```java
mMediaRecorder.setVideoFrameRate(20);
mMediaRecorder.setVideoEncodingBitRate(600 * 1000);
mMediaRecorder.setAudioEncodingBitRate(128 * 1000);
```

**iOS Support**

I'm not a iOS developer, so I wouldn't be able to help in here, but if someone would like to, I would be very happy to provide an Android implementation. 😉

sibelius, 2018-01-07:
this 3 new props looks a good ideia

feel free to send a PR and we'll help you

n1ru4l, 2018-12-19:
@djleonskennedy Please feel free to send a pull request that implements these features.

closed 2019-06-08 by stale[bot]