react-native-camera/react-native-camera #689: Feature request: Real-time Preview Processing

kayazinc, 2017-05-04:
### Expected behaviour
#### Use-case
While in camera preview mode, it is able to detect object such as "Rectangle" or "Facial" and draw a rectangle around the object.

Is it possible to achieve the use-case in javascript, such as having a new onPreivew() method that returns the image and then perform the detection algorithm on the image?

Thanks, any advice are welomed.

dotnetprofessional, 2017-05-10:
I'd like to add to this - I'm currently attempting to add an event that returns the preview image so that additional processing can be done. Taking captures is just too slow on Android (iOS is fairly quick). On Android this looks to be fairly straightforward as the byte[] is available, but for iOS its seems a bit more involved. Though any pointers are welcome, as Obj-C isn't a language I know :)

npomfret, 2017-05-10:
@dotnetprofessional  Have you tried master?  There are some changes which may improve capture performance on android that I don't think have been released yet.

The iOS version of the android byte[] is [here](https://github.com/lwansbrough/react-native-camera/blob/master/ios/RCTCameraManager.m#L590) i think.

dotnetprofessional, 2017-05-13:
@npomfret Thanks for the details - I've not tried master. I took a look at the iOS link you gave me, but that's if you do a capture. I'm looking at getting the byte[] from a preview so that its really fast. My scenario (which is odd I'll admit) is to capture the video preview and save the image as a JPG so that a webview can keep refreshing it. The idea is to provide a pysdo stream within a webview ie a web page. We can use WebRTC on Android but iOS doesn't support that.

barron9, 2017-05-16:
please take a look at anyline ocr react-native demo for this

dotnetprofessional, 2017-05-16:
@barron9 Thanks but the pricing model would bankrupt us given our product.

npomfret, 2017-05-17:
What are the rules regarding plugging commercial products on here?

barron9, 2017-05-17:
no just take the idea, their custom camera draws rectangles around objects words etc in every 10 or 20 ms. to recognize cam refreshes itself so processes while on preview

sibelius, 2017-10-27:
@dbishai is this https://github.com/dbishai/react-native-camera/commit/75553a6a1d09842558ad16fd6cb16abcd67cdcad performa performatic?

dbishai, 2017-10-31:
@sibelius, what does "performa performatic" mean?

ducpt2, 2017-12-13:
+ 1.

closed 2019-06-08 by stale[bot]

rafaelcavalcante, 2020-09-29:
This should be reopened

Related: #2984