react-native-camera/react-native-camera #696: android 6.0  no camera found

wq13718048209, 2017-05-08:
Dear author：
  Camera on the ios is normal, but on android 6.0 phone does not open, tip me no camera found, but the example of you on my mobile phone can run normally, permissions are added, who know what the reason

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**:
- **React Native version**:0.43
- **React Native platform + platform version**: iOS 9.0, Android 6.0, etc

### react-native-camera
**Version**: npm version or "master"

gsteyn, 2017-05-09:
I'm having the same issue. When opening the scene with the Camera component added, the emulator/device shows a blank screen, and when tapping the screen the app crashes with no error message.

It would seem that react-native-camera doesn't work on Android 6 or higher?

React Native version: 0.41.2
Android SDK: 23
react-native-camera version: 0.6

npomfret, 2017-05-10:
I have it working on android 6.  Maybe its something to do with your device?  Are you seeing any exceptions in Log cat?

gsteyn, 2017-05-11:
Hi @npomfret,

I found the problem. There were some permissions missing in the manifest and the app's permissions also needed to be set on the device to allow it to use the camera.

closed 2017-05-12 by npomfret

Graren, 2017-05-25:
Hi! could you show which permissions did you need to set? I apologize for opening this

gsteyn, 2017-05-25:
Hi @Graren,

I added: 
` <uses-permission android:name="android.permission.CAMERA" />
 <uses-feature android:name="android.hardware.camera" />
 <uses-feature android:name="android.hardware.camera.autofocus" />`
to my `AndroidManifest.xml`.

Also note that you need to use a device. Using the emulator will not work.

I hope you come right!

Borisboky, 2017-05-26:
I'm also getting `no camera found` error and the @gsteyn solution didn't work for me.

The permissions which I added are : 

```
<uses-permission android:name="android.permission.CAMERA" />
<uses-feature android:name="android.hardware.camera" />
<uses-feature android:name="android.hardware.camera.autofocus" />

<uses-permission android:name="android.permission.RECORD_AUDIO"/>
<uses-permission android:name="android.permission.RECORD_VIDEO"/>
<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
```

Any advice?

gsteyn, 2017-05-26:
Hi @Borisboky,

I'm assuming you're using a device, right? If so, then what could be causing this problem is permissions in your device itself. I had to go to `Settings > Apps > MyApp > Permissions` and then enable Camera permissions.

It seems that you only need to do this once, unless you uninstall the app, in which case you'll have to give permissions again.

I'm unsure as to why the permissions aren't automatically set in the app's permissions settings, as they're specified in the manifest file.

Borisboky, 2017-05-26:
Thanks @gsteyn.  Strange that those permissions aren't automatically set, but that did the trick.

gsteyn, 2017-05-26:
Glad I could help @Borisboky. Yeah I found it quite strange too, although I didn't investigate it further... deadlines are sometimes greater than curiosity ;)

grigored, 2017-07-16:
Any ideas as to why we have to manually request camera permissions in Settings -> Apps?

furedal, 2017-08-01:
@grigored
Since Android 6.0 (API level 23), it is required by the developer to request permission before using some futures. Managing Camera and External Storage are two of them.
Facebook provides you with an example:
[https://facebook.github.io/react-native/docs/permissionsandroid.html](url)

Read more about Runtime permissions:
[https://developer.android.com/training/permissions/requesting.html](url)

shubhangi-asgola, 2017-11-07:
@npomfret 
I am still not able to get it working. I am using
 react-native-camera@0.6
targetSdkVersion 22 (which does not require to add permissions explicitly. I have added permissions in AndroidManifest.xml)
react-native@0.42

It works well in iOS and android emulator, however app crashes in android device(release mode APK). If I run it in debug mode in device it gives no camera  found error. I don't know where I am going wrong.